OllisGit/OctoPrint-PrintJobHistory #2: Plugin manager states update is available but it will not install

joeySeal, 2019-11-04:
Octprint update manager will not update to rc2

`2019-11-03 19:33:02,261 - octoprint.plugins.softwareupdate - INFO - Starting update of PrintJobHistory to 1.0.0-rc2...
2019-11-03 19:33:39,160 - octoprint.plugins.softwareupdate - INFO - Update of PrintJobHistory to 1.0.0-rc2 successful!
2019-11-03 19:33:39,345 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2019-11-03 19:33:39,348 - octoprint.plugins.softwareupdate - INFO - Restarting...
2019-11-03 19:33:39,538 - octoprint.server - INFO - Shutting down...
2019-11-03 19:33:39,682 - octoprint.events - INFO - Processing shutdown event, this will be our`
[image]

OllisGit, 2019-11-04:
Hi @joeySeal ,

did you already try:
- uninstall RC1
- reboot system
- install RC2

If this also not working, please attach the full octoprint.log file.

joeySeal, 2019-11-04:
@OllisGit how should I be installing? Via URL? (https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/latest/download/master.zip) or from the plugin manager. It does not appear in the plugin repo search.

Putzi76, 2019-11-04:
> Hi @joeySeal ,
> 
> did you already try:
> 
> * uninstall RC1
> * reboot system
> * install RC2
> 
> If this also not working, please attach the full octoprint.log file.

it works for me.
Thanks

OllisGit, 2019-11-04:
@joeySeal 
URL as mentioned in Homepage: https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/latest/download/master.zip

karnwatcharasupat, 2019-11-07:
@OllisGit I just looked at `setup.py` of rc2 and it seems that the plugin version is not updated

```
# The plugin's version. Can be overwritten within OctoPrint's internal data via __plugin_version__ in the plugin module
plugin_version = "1.0.0rc1.1"
```

OllisGit, 2019-11-07:
Hi @joeySeal ,
Damm, I forgot to push the setup file with the new version-number.

If you did a clean installation, you should receive rc2 (only with wrong version number), but updating from rc1.1 to rc2 could not work.

I will improve my build/releasmanagement, so that the setup version-number is the same as the git release-number.

Now rc2.1 is out, please test and give feedback.

Thx, Olli

b-morgan, 2019-11-07:
I have rc2.1 installed but I still get an update available message, see attached.
[image]
If there is something else you need from me, let me know.

b-morgan, 2019-11-07:
I notice from the above screenshot that the name of the plugin in the manager is "Job History" but the project is PrintJobHistory. Should the two be more similar?

OllisGit, 2019-11-07:
Hi @b-morgan ,
thanks a lot for pointing it out to me.

1) You installed the current version "rc2.1" and in the right popup the previous plugin "rc2" is mentioned. I think Octoprint has some "challenges" to identify the right number, so I removed the old "rc2" from my github release-panel. That should solve the issue.
2) I am not really "lucky" with the name "PrintJobHistory", so the idea was to shorten that Name and reuse the internal PluginId, but that is confusing. I will change that in the next version.

Again, thanks for your testing support!
Olli

commit 6bdedeb7 (2019-11-15): RC3
fix #2, renaming of Pluginname
fix #5, csv-export of accent characters
improved #4, text in plugin dependency check
closed 2019-11-15 by OllisGit