OllisGit/OctoPrint-PrintJobHistory #3: Completed prints do not show up on the history

karnwatcharasupat, 2019-11-04:
I have tried printing two jobs but unfortunately, none of them shows up in the history.

[image]

OllisGit, 2019-11-04:
Hi @karnwatcharasupat ,
hmmm...no idea!

- Please attach the octoprint.log (download via logging menu)
- Please look into the plugin settings in which folder the database was installed and then check if the database file is really there.

karnwatcharasupat, 2019-11-06:
Hi, seems like my octoprint log still shows the plugin as rc1 despite me updating it. I will have to try uninstalling it completely and install rc2 from the url.

OllisGit, 2019-11-07:
Hi @karnwatcharasupat ,
Damm, I forgot to push the setup file with the new version-number.

If you did a clean installation, you should receive rc2 (only with wrong version number), but updating from rc1.1 to rc2 could not work.

I will improve my build/releasmanagement, so that the setup version-number is the same as the git release-number.

Now rc2.1 is out, please test and give feedback.

Thx, Olli

karnwatcharasupat, 2019-11-08:
I just installed rc2.1 (uninstalled rc1 and did a clean install). Printing a test print now. I'll update you soon.

karnwatcharasupat, 2019-11-08:
Hi. Unfortunately the print still doesn't show up on history. The log file also somehow show the ver as rc1.1 despite a clean install. The db file was at the directory it should be.

[octoprint (4).log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/3823810/octoprint.4.log)

OllisGit, 2019-11-08:
Hi @karnwatcharasupat,
thx, for providing the log-file really helpful!

As you already mentioned the old version is still installed and also there is something wrong with the filamentmanager-plugin:
```
2019-11-08 04:05:29,312 - octoprint.plugins.filamentmanager - ERROR - Failed to fetch profiles lastmodified timestamp: 'NoneType' object has no attribute 'get_profiles_lastmodified'
```
You should try to reinstall the plugin. 
- via plugin manager
- you need to cleanup the config.yaml fragments and also the PrintJobHistory-Data folder (where the database is located). This is a bug in OctorPrint and should be sovled in one of the next version. See: https://github.com/foosel/OctoPrint/issues/3299

After restarting and reinstalling the plugin with the URL from my homepage, you can test the functionality really fast: start a print, cancel the print immediately. After a couple of seconds the edit Job-Dialog should appear and the job item is listed in the history table.

To reduce the log-volume you could also disable the "Telegram-Plugin", because it produce a lot of error entries.

Well it is only a release candidate, but this is annoying, sorry about that!

Thx, for testing
Olli

OllisGit, 2019-11-15:
Hi @karnwatcharasupat ,
I released a new version (rc3). 

Please test again and be aware of the plugin name change.

Thx, again
Olli

Quintin9801, 2019-11-21:
I am having this same problem, I run prints that are often 60+ hours and using this plugin would be great for comparing settings times! I can't wait for the full version of this awesome plugin!

EddyMI3d, 2019-11-22:
Hello @OllisGit !
I've got it installed the last days (rc3). The first print after installation showed up in the list, but I was not able to change the thumbnail image. 
When I booted the RPi the next day, the list was empty again. New prints never showed up in the list.
I checked the database location : `pi@octopi:~/.octoprint/data/PrintJobHistory`
The file `printJobHistory.db` is not empty.

OllisGit, 2019-11-23:
Hi @Quintin9801 @EddyMI3d ,
I found a formatting issue the last days maybe your issue is related to that.

Please do the following:
- start octoprint server
- open the browser
- switch to the PrintHistory-Tab
- open the settings and navigate to logging and download octoprint.log

if the log file is to big e.g. >20MB please delete the log-file, reload the PrintHistory-Tab and then download the log.file and attach it to this issue.

EddyMI3d, 2019-11-23:
Ok, here is mine:
[octoprint.log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/3882367/octoprint.log)

OllisGit, 2019-11-24:
@EddyMI3d  thanks for the log!
The big should be fixed in the newest version RC4.

@EddyMI3d @Quintin9801 @karnwatcharasupat Please test and give me a feedback!

Thx, in advance
Olli

EddyMI3d, 2019-11-24:
Hello Olli!
I just give it a shot... 

Thank you for your efforts

OllisGit, 2019-12-03:
Hi @EddyMI3d ,
damn...looks like the formatting isssue is still there.
Please so kind and attach your database-file to this issue (could be done via download-button in settings-dialog).

EddyMI3d, 2019-12-04:
Hello Olli!

I just tried it on my second system. Same issues, the difference is, that I get an error message .
I packed the files together in a zip file
[PrintJobHistory.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/3922014/PrintJobHistory.zip)

ThomasTL, 2019-12-05:
I believe i fixed your float formatting issue @OllisGit 
Please check out this pull request: 
https://github.com/OllisGit/OctoPrint-PrintJobHistory/pull/9

OllisGit, 2019-12-06:
Hi @ThomasTL thanks a lot for your pull-request. It will be included in the next version (hopefully this weekend)

OllisGit, 2019-12-07:
Hi @ThomasTL , @EddyMI3d , @Quintin9801 , @karnwatcharasupat,
new (and last) release candidate (RC5 )is out and the issue should be solved.

Please test and give me a feedback.
Thx, in advance
Olli

EddyMI3d, 2019-12-07:
Hi Olli!

I just installed it. There former prints are still there, but without an image.
A quick print also includes an image.

It seems to work now! Great!

OllisGit, 2019-12-08:
Hi @EddyMI3d ,
to get back the images please do the following:
- Look into the "snapshot-file location", see Plugin-Settings "Storage". This folder includes all taken pictures. Filename is the startprinttime.
- If you manually save a printjob in the previous plugin version, the startprinttime was modified (seconds were cut)
- Please rename the missing pictures jpg files to something like this: ```20191207-174900.jpg```, because in the database the seconds will be also ```00```.
- After that you can reload the table and the pictures should be appear again.

BR
Olli

EddyMI3d, 2019-12-08:
Haello @OllisGit!

I'll will do that ASAP. Thank you for that info!

Ewald

closed 2020-05-20 by OllisGit

johnbozuk, 2021-02-21:
Sadly none of my print jobs save either if completed or failed, I have now manually added one print job to see if it starts to work,

dbellizzi, 2021-03-21:
I'm now getting:
```
2021-03-20 11:16:10,150 - octoprint.plugin - ERROR - Error while calling plugin PrintJobHistory
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 614, in on_event
    self._printJobFinished("success", payload)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 436, in _printJobFinished
    self._createAndAssignFilamentModel(self._currentPrintJobModel, payload)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 254, in _createAndAssignFilamentModel
    filemanentModel.usedLength = self._filamentManagerPluginImplementation.filamentOdometer.totalExtrusion[0]
AttributeError: 'FilamentManagerPlugin' object has no attribute 'filamentOdometer'
```

I'm running `Filament Manager (1.7.2)` and `PrintJobHistory (1.10.0)`

Delarado, 2021-03-22:
I have the same issue... Same versions too! Mine is broken since December last year (thats the last job captured)
Has there been an update that caused an issue?

2021-03-22 20:38:04,246 - octoprint.plugins.PrintJobHistory - INFO - Start capturing print job
2021-03-22 20:38:04,259 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Start parsing Slicer-Settings
2021-03-22 20:38:04,466 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Finished parsing Slicer-Settings
2021-03-22 20:38:04,470 - octoprint.plugins.PrintJobHistory - WARNING - Thumbnail not found in print metadata
2021-03-22 20:38:04,471 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Try taking snapshot '/home/pi/.octoprint/data/PrintJobHistory/snapshots/20210322-203754.jpg' from 'http://127.0.0.1:8080/?action=snapshot'
2021-03-22 20:38:04,474 - octoprint.plugin - ERROR - Error while calling plugin PrintJobHistory
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 620, in on_event
    self._printJobFinished("canceled", payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 436, in _printJobFinished
    self._createAndAssignFilamentModel(self._currentPrintJobModel, payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 254, in _createAndAssignFilamentModel
    filemanentModel.usedLength = self._filamentManagerPluginImplementation.filamentOdometer.totalExtrusion[0]
AttributeError: 'FilamentManagerPlugin' object has no attribute 'filamentOdometer'
2021-03-22 20:38:04,476 - octoprint.plugins.filamentmanager - INFO - Updating Filament usage for octoprint configured toolcount: 1
2021-03-22 20:38:04,476 - octoprint.plugins.filamentmanager - INFO - Filament tracked toolcount: 1
2021-03-22 20:38:04,477 - octoprint.plugins.filamentmanager - INFO - Filament tracked values: [0.0]
2021-03-22 20:38:04,480 - octoprint.plugins.filamentmanager - INFO - Filament used: 0.0 mm (tool0)
2021-03-22 20:38:04,514 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Process snapshot image

Delarado, 2021-03-22:
https://github.com/OllisGit/OctoPrint-FilamentManager/releases

Says it doesn't work with print job history.  Looks like 1.7.0 might be the one to go with..

Delarado, 2021-03-22:
Downgrade to FilamentManager 1.7.0 has fixed for me.
From the plugin screen in Octoprint:
Get More
From URL
Paste in: https://github.com/OllisGit/OctoPrint-FilamentManager/releases/download/1.7.0/master.zip

Complete that, restart, and you should be good to go again!