OllisGit/OctoPrint-PrintJobHistory #4: Difference between not-installed and installed but disabled

b-morgan, 2019-11-07:
I'm not sure if you can tell the difference but it took me a while to discover that DisplayLayerProgress was installed but it was disabled. If you can tell the difference, changing the message would be helpful.

OllisGit, 2019-11-07:
Hi @b-morgan ,

the new Dialog-Text will look like this in the next version:
[image]

b-morgan, 2019-11-07:
Perfect!

OllisGit, 2019-11-15:
Hi @b-morgan ,
new version rc3 is out. 
If you want to test...or if you have the time for a poll: ["What is the next big thing?"](https://github.com/OllisGit/OctoPrint-PrintJobHistory/issues/6 ) 

Thx,
Olli

closed 2019-11-24 by OllisGit