OllisGit/OctoPrint-PrintJobHistory #6: [Poll] What is the next big thing?

OllisGit, 2019-11-15:
You can influence the future development of this plugin!

Please vote for the feature you would like to see in the next version.
If your feature is not an option in the below poll, please add a comment (one comment for one feature) and then other users could vote via reaction-icon (Thumbs-Up/Down).

[image]
[image]
[image]
[image]
[image]
[image]
[image]

3Simplex, 2019-12-06:
I would like the user identified for each print job, and the ability to list jobs by user.

ThomasTL, 2019-12-06:
When installing the plugin, the ability to pre-load the DB with past print done based on the files already downloaded in Octopi.
Since it is possible on Octopi to see for each file uploaded, the one printed failed or successful with the last print time when it is successful, it could be very nice to pre-load this information in the DB when installing the plugin. 
A kind of partially restore previous print history.

OllisGit, 2019-12-07:
Hi @ThomasTL,
technically it should be possible, because everything is stored in several ```.metadata.json``` files:
[image]
From my point of view the "Importer" needs a UI where you can see all prev. jobs and then you can select each job you want to import to the printjob-database. After that, you can edit each job individually (as normal).
Developing UIs takes a lot of time..... so personally it is not in my top5-features I would like to implement next, but I will take it to the poll and to my planing board: https://github.com/OllisGit/OctoPrint-PrintJobHistory/projects/1

OllisGit, 2019-12-07:
Hi @3Simplex ,
your request is part of "Add more filter options to table...". 
The Username is definitely one attribute that is part of this feature.

ThomasTL, 2019-12-08:
Sure. 
@OllisGit maybe i'll have a look at it then.

Another feature that could be nice, is to have an option allowing to display the table rows in a condensed manner rather than having all the information in it. Then be able to view the print details when clicking on the row or on the edit button.

jto2000, 2019-12-10:
Hi,

I was thinking to a new point: As the big point of the pole is having stats / diagram and so on, maybe it could be great to implement some "maintenance" task, as changing hotend, unclogging  etc... 
This could allow to identify preventive action, or to follow the eventual improvement of the printer setup.
Hope it will speak to somebody ;)

giacomo30196, 2020-01-06:
Hi 
i'm thinking of a new function. it would be cool, once a print is finished, that the plugin automatically log data to a new row of a google sheet! i'd like to implement it myself, but i'm completely out of clue...

OllisGit, 2020-01-06:
Hi @jto2000,
I was thinking about implementing "tags-inputfields", like in my SpoolManager-Plugin, see https://github.com/OllisGit/OctoPrint-SpoolManager/blob/master/screenshots/editSpool-dialog.png
With this feature you can create "labels" as many you like and of course create later some stats.

OllisGit, 2020-01-06:
Hi @jto2000 ,
nice idea....but why an other "storage". Sorry, but I think I will not implement such feature in this plugin.
But like my other Plugin "DisplayLayerProgress" I will send a Custom-Event to the Eventbus after a history-entry is created, so each other Plugin-Developer could listen for such event and send the history-info where ever he/she wants.

Do you have programming experience (no matter what language) or willing to learn? I can "teach" you a little bit. Your plugin-request should be easy to implement....I think!

giacomo30196, 2020-01-06:
HI @OllisGit, thank you for the reply!. i'll take a look at yoru displaylayerprogress plugin and see if i manage to understand something. I have never studied programming language but i'm able to code in arduino/C, python, js at very basic level because i learnt from examples... i'll try to make something and than i'll update you! thank you!

giacomo30196, 2020-01-09:
@OllisGit Hi, i actually made it, after reading your code as suggest i understood and started from scratch. As you said it has been quite easy. I'm planning to make it ui-configurable in some time... maybe after passing my university exams... thank you btw!

mariowitdoek, 2020-05-10:
Hi Olli, 
if you implement power consumption (please via standard or vendor unlocked protocol || kind of plugin requester) please add the possibility to add "an electricity bill cost" ....
I just found your history plugin and will test/implement it: KUTGW!

Mario

PS I have a friend-builder which has also the 'olli' brandname :p

OllisGit, 2020-05-11:
Hi @mariowitdoek ,

I am not sure what you mean with "standard" or "vendor unlock protocol".
My current assumption is that I can grab the values from other plugins like "tasmota" or "TP-LinkSmartplug"
Or an other approach is to call directly a REST-API on the SmartPlug to get the consumption.

rongith, 2020-06-08:
I don't find a way to change the status of a print from success to failed. Then I supposed it is not implemented. And I'm not sure that the 3rd line in your poll is about that.
Then I would like to change the status a particular print.

6ffm70, 2020-06-14:
> Hi @mariowitdoek ,
> 
> I am not sure what you mean with "standard" or "vendor unlock protocol".
> My current assumption is that I can grab the values from other plugins like "tasmota" or "TP-LinkSmartplug"
> Or an other approach is to call directly a REST-API on the SmartPlug to get the consumption.

Hi @OllisGit 
I'm not proficient in coding, but I've found an example of an xml message a smartplug (_Fritz!DECT 200_) returns: https://avm.de/fileadmin/user_upload/Global/Service/Schnittstellen/AHA-HTTP-Interface.pdf (page 11) as an example. It returns 
```
<powermeter>
<power>0</power><energy>707</energy><voltage>230252</voltage>
 </powermeter>
```
for power consumption, and 
```
<temperature><celsius>288</celsius><offset>0</offset></temperature>
```
for temperature.

There's some more documents here [https://avm.de/service/schnittstellen/](url); possibly also how to connect to the router which has the interface to the smartplug.

OllisGit, 2020-06-21:
Hi @6ffm70,
thx for the info. I added this to my Backlog-Card.

6ffm70, 2020-07-28:
> Hi @6ffm70,
> thx for the info. I added this to my Backlog-Card.

Hello Olli, there's an API written in js that may be a little closer to implementation for that particular plug: https://github.com/andig/fritzapi

gdcircuit, 2020-10-31:
Requesting support for simplify3d slicer settings capture. Simply3d does not use an =.