OllisGit/OctoPrint-PrintJobHistory #7: Update Available for previous version (how to make it go away)

b-morgan, 2019-11-15:
I have updated to 1.0.0rc3 (through the Update Available) but after reloading, I still have an Update Available for v1.0.0-rc2.1.

Please tell me what I have to do to get this plugin in sync with itself. I'm guessing a complete removal of all traces followed by a fresh install of v1.0.0-rc3 but I'm not sure exactly what "complete removal" requires.

OllisGit, 2019-11-15:
hmmm...I think OctoPrint has some problems with the "rc" postfix.
Uninstall should not be necessary, because the plugin-id was not changed.

I removed the rc2.1 from github, so it is not visible for octoprint anymore.

Just for my understanding: 
Which version is listed in Software Update list? 
"Job History rc2.1" or "PrintJobHistory rc3" or both?

Maybe the release cache is out of sync. please force a lookup. click on advanced option and then the button in the bottom.
[image]

b-morgan, 2019-11-15:
Only PrintJobHistory (1.0.0rc3) is listed.

The Force check for update cleared the Update Available for 1.0.0rc2.1

OllisGit, 2019-11-15:
puhh...okay! Thx, for your fast feedback!

closed 2019-11-15 by OllisGit