OllisGit/OctoPrint-PrintJobHistory #8: Username is incorrect incase of using an api key

phschoen, 2019-11-30:
If the user is generating an api key the user name in the db is "_api"
This does not help to identify the user if there are multiple ones and all are printing with the api key.

therefore it would be nice to have also something like username and via (web/api/telegram?)

OllisGit, 2019-12-03:
Hi @phschoen !
Thx, for the hint! Unfortunately I have no idea how I can receive the associated displayname of the user. The event-payload only include the username...and this is ```_api``` when starting via API.

I ask in the octoprint-form for help, maybe I receive a valid answer.
https://community.octoprint.org/t/username-of-started-printjob-via-api/14000

BR
Olli

OllisGit, 2019-12-06:
Hi @phschoen ,

please read the answer from the forum.

Short: 
1) create a dedicated user (Settings/Access Control)
2) login as the new user
3) In user settings create a new Application Key (don't forget to enter a Application identifier)

After that you can use this api-key to start a print and then the associated username will be appear in the history.

phschoen, 2019-12-09:
Thanks for the feedback. :+1: with a fresh api key this works as you sad. 
with the telegram app the user is `-` and therefore you also don't see the username with `"owner"`. So this can be only fixed within the telegram app. 

An other thing which would be an improvement however is that non admin users can delete any history entries. This should only be possible for the admin user itself.

OllisGit, 2019-12-09:
Hi @phschoen !
Yap, security-topics were not implemented yet...it's on my list.

closed 2019-12-09 by phschoen