OllisGit/OctoPrint-PrintJobHistory #10: Only Failed prints have a picture

RomRider, 2019-12-05:
Hi,

Thanks for this amazing plugin, it's nice to be able to track all the prints!
It seems currently that only the failed prints have their snapshot saved.
Actually, when the prints finishes, I can see the picture in the popup, then I save the print in the history. After that, while browsing the history, the picture will not be displayed.

Do you need anything to help you debug this?

OllisGit, 2019-12-06:
@RomRider thanks for testing the release candidate.
I analysed the error and it will be fixed in the next version (hopefully this weekend)

BR
Olli

OllisGit, 2019-12-07:
Hi @RomRider,
new (and last) release candidate (RC5 )is out and the issue should be solved.

Please test and give me a feedback.
Thx, in advance
Olli

OllisGit, 2019-12-08:
Hi @RomRider ,
to get back the images please do the following:
- Look into the "snapshot-file location", see Plugin-Settings "Storage". This folder includes all taken pictures. Filename is the startprinttime.
- If you manually save a printjob with the previous plugin version (RC4), the startprinttime was modified (seconds were cut)
- Please rename the missing pictures jpg files to something like this: ```20191207-174900.jpg```, because in the database the seconds will be also ```00```.
- After that you can reload the table and the pictures should be appear again.

BR
Olli

OllisGit, 2020-01-26:
Hi @RomRider ,

could you retest the behaviour?

RomRider, 2020-01-26:
Sorry, I didn't report back but it works fine with the latest version, thanks a lot!

closed 2020-01-26 by RomRider