OllisGit/OctoPrint-PrintJobHistory #11: Enhancement; Import old PrintHistory data

b-morgan, 2019-12-08:
I ran the old Print History plugin for a quite a while so there is a fair bit of historical data. Would it be possible to import that data. 

A general import to go with your export in the plugin and an external program to do the conversion of the old data to CSV would be acceptable.

OllisGit, 2019-12-08:
Hi @b-morgan ... I was waiting for this feature...I think I combine this with the ```.metadata.json``` importer from #6. 
Should be easy to implement.

I am not sure in which version this will be implemented.

OllisGit, 2020-05-03:
Hi @b-morgan,
new RC6 is out, BUT I need to disable the Import-Feature. 
https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.0.0rc6
I didn't have much time to test everything.

In the final 1.0.0 release the importer is a "must-have"

OllisGit, 2020-06-06:
Hi @b-morgan 
in the lasted version 1.3.0 there is an CSV exporter for all PrintHistory-Plugin data.

Please test and give feedback.

closed 2020-09-09 by stale[bot]