OllisGit/OctoPrint-PrintJobHistory #13: Prints Requiring Multiple Rolls of Filament Show Used Filament Incorrectly

Quintin9801, 2019-12-23:
When printing large objects that require multiple rolls of filament, The filament sensor on the machine is triggered and sends an M600 command to the printer and Octoprint to change filament, as well as a pause command for Octoprint.

Currently, the amount of filament on for multiple roll prints only counts the amount used until the filament is changed after the filament change the amount is not recorded in the history.

Currently, I do not have any GCODE scripts that execute before or after the print is paused because it is all handled by the printer's firmware.

[image]
[image]

Additional Info:
Octoprint Version: 1.3.12 
PrintJobHistory Version: RC5
Printer: Modix Big 60 w/MKS Gen 1.4
Firmware: Marlin 1.1.9
Octoprint Server OS; Windows 10

OllisGit, 2019-12-23:
Hi @Quintin9801 ,
first: Wow!!! 3 days and 10 our print, respect! What a cool big printer ;-)

After the print I am reading the values from the "FilamentManager-Plugin" and I think this plugin doesn't support filament change, see Isssue: https://github.com/malnvenshorn/OctoPrint-FilamentManager/issues/76

But my assumption is, after the change the used weight should continue increasing and not stopping....or do you use more then one extruder? Because currently I only support one extruder.

I will take a look after x-mas.
Olli

Quintin9801, 2019-12-23:
@OllisGit 
Thanks for such a fast reply! To answer your question about the number of extruders, there is only one.

Have a merry x-mas sir,

Quintin

OllisGit, 2020-06-13:
Hi @Quintin9801!

fyi:
Well, it took a while.....but currently I am continuing developing my spool-manager-plugin  and as a reference I use the FilamentManager plugin. 

There is an issue after the printer paused, everything is reseted and I don't know why he implement this "feature".
The evaluated extrusion length should only be reset, if you change the spool and the current length value should assign to the last spool.

OllisGit, 2021-05-24:
Try to use the lastest PJH-Plugin V1.14.0 and SpoolManger V1.4.0, that should work!

Please test and give feedback!
Olli

OllisGit, 2021-10-30:
No feedback. If needed raise a new ticket.

closed 2021-10-30 by OllisGit