OllisGit/OctoPrint-PrintJobHistory #14: The plugin stopped recording print jobs a month ago. What do I need to do to troubleshoot that?

simpat1zq, 2020-01-20:
Like the subject says. And I don't know what I changed at that time since it's been so long. I only recently noticed it. I don't see any settings that stand out that could cause this. There are a large number of jobs recorded before that.

OllisGit, 2020-01-20:
Hi @simpat1zq!

- Which RC did you use?

- The Tab "Print Job History" is still visible, right?

- Did you already look into the ```octoprint.log``` (settings/Logging), maybe there is a error listed.

- Please check if the ```printJobHistory.db``` file is still available in this path:```<octoprint>/data/PrintJobHistory/printJobHistory.db``` . If yes, please attach this file to this issue.

simpat1zq, 2020-01-24:
I'm on the latest RC of the plugin. And that reminded my, I'm also on the latest RC of OctoPrint. That might be where the issue is. I believe the time the plugin stopped working is when I upgraded to RC1 of OP. Do you know if the plugin is working with that RC?

Here's the relevant section of the logs. I had wiped out the db, so I'm not sure if that would be any use to you. Let me know if you still want that. :

2020-01-24 07:48:49,503 - octoprint.plugins.DisplayLayerProgress - INFO - Printing stopped. Detailed progress stopped.
2020-01-24 07:48:49,507 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Try taking snapshot '/home/hass/.octoprint/data/PrintJobHistory/snapshots/20200124-074438.jpg' from 'http://127.0.0.1:8080/?action=snapshot'
2020-01-24 07:48:49,507 - octoprint.plugin - ERROR - Error while calling plugin PrintJobHistory
Traceback (most recent call last):
  File "/home/hass/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/hass/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 340, in on_event
    self._printJobFinished("canceled", payload)
  File "/home/hass/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 277, in _printJobFinished
    self._createAndAssignFilamentModel(self._currentPrintJobModel, payload)
  File "/home/hass/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 155, in _createAndAssignFilamentModel
    fileData = self._file_manager.get_metadata(payload["origin"], payload["file"])
KeyError: 'file'
2020-01-24 07:48:49,508 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Try converting thumbnail '/home/hass/.octoprint/data/PrintJobHistory/../UltimakerFormatPackage/CE5_Deco_OK.png' to '/home/hass/.octoprint/data/PrintJobHistory/snapshots/20200124-074438.jpg'
2020-01-24 07:48:49,552 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Converting successfull!
2020-01-24 07:48:49,759 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Process snapshot image
2020-01-24 07:48:49,912 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=PrintCancelled
2020-01-24 07:48:49,919 - octoprint.plugins.stats - INFO - Printer Stats - on_event
2020-01-24 07:48:49,936 - octoprint.plugins.DisplayLayerProgress - INFO - Printing stopped. Detailed progress stopped.
2020-01-24 07:48:49,939 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=PrintFailed
2020-01-24 07:48:49,940 - octoprint.plugins.stats - INFO - Printer Stats - on_event
2020-01-24 07:48:49,956 - octoprint.plugins.telegram - INFO - An Exception in get final time : unsupported type for timedelta seconds component: unicode
2020-01-24 07:48:49,957 - octoprint.plugins.telegram - INFO - Sending a message: Printing CE5_Deco_OK.ufp.gcode failed. with_image=False with_gif=True chatID= 297408197
2020-01-24 07:48:50,642 - octoprint.plugins.telegram - INFO - Will try to create a gif 
2020-01-24 07:48:50,647 - octoprint.plugins.telegram - INFO - will try to save image in path /home/hass/.octoprint/data/telegram/tmpgif
2020-01-24 07:48:56,948 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Operational"
2020-01-24 07:49:31,305 - octoprint.plugins.filamentmanager - INFO - Filament used: 0.0 mm (tool0)
2020-01-24 07:50:24,617 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-01-24 07:58:24,857 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.45.87
2020-01-24 07:58:24,906 - octoprint.server.util.flask - INFO - Passively logging in user _api from 192.168.45.87
2020-01-24 07:58:26,913 - octoprint.plugins.PrintJobHistory - INFO - Plugin-State: PreHeat=enabled DisplayLayerProgress=enabled filamentmanager=enabled ultimakerformat=enabled
2020-01-24 08:00:00,149 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.45.87
2020-01-24 08:05:09,966 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('SELECT "t1"."databaseId", "t1"."created", "t1"."userName", "t1"."fileOrigin", "t1"."fileName", "t1"."filePathName", "t1"."fileSize", "t1"."printStartDateTime", "t1"."printEndDateTime", "t1"."duration", "t1"."printStatusResult", "t1"."noteText", "t1"."noteDeltaFormat", "t1"."noteHtml", "t1"."printedLayers", "t1"."printedHeight" FROM "pjh_printjobmodel" AS "t1" ORDER BY "t1"."printStartDateTime" DESC LIMIT ? OFFSET ?', [10, 0])
2020-01-24 08:05:09,967 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('SELECT COUNT(1) FROM (SELECT 1 FROM "pjh_printjobmodel" AS "t1") AS "_wrapped"', [])
2020-01-24 08:05:11,239 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('SELECT "t1"."databaseId", "t1"."created", "t1"."userName", "t1"."fileOrigin", "t1"."fileName", "t1"."filePathName", "t1"."fileSize", "t1"."printStartDateTime", "t1"."printEndDateTime", "t1"."duration", "t1"."printStatusResult", "t1"."noteText", "t1"."noteDeltaFormat", "t1"."noteHtml", "t1"."printedLayers", "t1"."printedHeight" FROM "pjh_printjobmodel" AS "t1" ORDER BY "t1"."printStartDateTime" DESC LIMIT ? OFFSET ?', [10, 0])
2020-01-24 08:05:11,240 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('SELECT COUNT(1) FROM (SELECT 1 FROM "pjh_printjobmodel" AS "t1") AS "_wrapped"', [])
[... cut ...]
2020-01-24 08:05:47,682 - octoprint.server.util.flask - INFO - Passively logging in user simit from 192.168.45.61
2020-01-24 08:05:47,892 - octoprint.server.util.sockjs - INFO - User simit logged in on the socket from client 192.168.45.61
2020-01-24 08:05:49,655 - octoprint.plugins.PrintJobHistory - INFO - Plugin-State: PreHeat=enabled DisplayLayerProgress=enabled filamentmanager=enabled ultimakerformat=enabled
2020-01-24 08:06:08,288 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.45.61

OllisGit, 2020-01-26:
Hi @simpat1zq,
you are absolutely right. My RC work currently only with version 1.3.x.
The next version will be compatible with 1.4.x.
Please be patient.

simpat1zq, 2020-01-26:
Got it. Thanks for the help. Looking forward to it.

rprb93, 2020-03-10:
I have the same problem, but I think the problem is different.

`2020-03-10 21:25:24,416 - octoprint.plugins.PrintJobHistory - INFO - Start initializing
2020-03-10 21:25:24,418 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Init DatabaseManager
2020-03-10 21:25:24,422 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Creating database in: /home/pi/.octoprint/data/PrintJobHistory/printJobHistory.db
2020-03-10 21:25:24,424 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Check if database-scheme upgrade needed.
2020-03-10 21:25:24,432 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('SELECT "t1"."databaseId", "t1"."created", "t1"."key", "t1"."value" FROM "pjh_pluginmetadatamodel" AS "t1" WHERE ("t1"."key" = ?) LIMIT ? OFFSET ?', [u'databaseSchemeVersion', 1, 0])
2020-03-10 21:25:24,462 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Done DatabaseManager
2020-03-10 21:25:24,464 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Init CameraManager
2020-03-10 21:25:24,465 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Snapshot-Folderr:/home/pi/.octoprint/data/PrintJobHistory/snapshots
2020-03-10 21:25:24,468 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Done CameraMenager
2020-03-10 21:25:24,473 - octoprint.plugin.core - ERROR - Exception while initializing plugin PrintJobHistory, disabling it
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 1469, in initialize_implementation
    implementation.initialize()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 61, in initialize
    self._settings.set( [SettingsKeys.SETTINGS_KEY_DATABASE_PATH], self._databaseManager.getDatabaseFileLocation())
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 524, in _func
    return orig_func(*args_mapper(args), **kwargs_mapper(kwargs))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/settings.py", line 1724, in set
    chain.set_by_path(path, value)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/settings.py", line 545, in set_by_path
    raise KeyError(key)
KeyError: 'PrintJobHistory'
2020-03-10 21:25:24,682 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk`

OllisGit, 2020-05-03:
Hi,
it's fixed in the new release candidate, see release description: https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.0.0rc6

closed 2020-05-03 by OllisGit