OllisGit/OctoPrint-PrintJobHistory #15: Small "funny" typo ;-)

BerndJM, 2020-01-22:
Correct List view:
[image]

And than ... the edit view:
[image]

Btw. it would be nice to see also the used weight in the list view, because this is IMHO the more common measurement and fits better to the information from the filament manager plugin.
Just my 2¢ ;-)

OllisGit, 2020-01-26:
Hi @BerndJM ,
thanks a lot for your 2¢ !!

Will be fixed in the next release.

Please be patient.
Olli

OllisGit, 2020-02-02:
In the next release the ui-output will be "m" and in the csv-output/input will be "mm".

OllisGit, 2020-05-03:
Hi,
it's fixed in the new release candidate, see release description: https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.0.0rc6

closed 2020-05-03 by OllisGit