OllisGit/OctoPrint-PrintJobHistory #19: Can No Longer Print Directly from Cura

BzowK, 2020-02-01:
Good Morning -

Installed RC5 yesterday after you suggested it to me on forum.  Once done (and after installing Preheat + Ultamaker dependencies), it works well but I'm having an issue.

Typically, I slice within Cura then choose to print via OctoPrint via it's Cura plug-in.  In the past, this loaded the gcode in OctoPrint and started the print immediately - allowing limited monitoring within Cura.  After installing your plug-in, choosing to print within Cura switches to the Monitor tab as usual + uploads the gcode, but doesn't load or start the print within OctoPrint.

To note, I'm using the latest OctoPrint as of writing.  Happy to test or submit anything needed. 

PS: Suggest options within settings to enable/disable features like post print report not only in web interface, but also separately for TouchUI interface.  Personally, I like that unexpected feature in web on PC, but rather not have it on touch screen.

Thanks!

OllisGit, 2020-02-02:
Hi @BzowK ,

with latest OP you mean 1.3.x (Python2), right and not 1.4.x!?!
Did you see any errors in the octoprint.log?
If you are not "familiar" reading the log, you could attach it to this issue.

I didn't get your suggestion. From my understanding you want the Edit-Job-Dialog should not appear on TouchUI after a print, only in Desktop Version and this behaviour should be switchable (enabled/disabled) !?!

BzowK, 2020-02-02:
Good Afternoon - 

Guess i should have been more specific :).  Yea, I'm running OC 1.3.12.  Now that I think about it, it could be due to the dependency plugins I had to install as well.  Anyways, with debugging on, my logs were over 200mb each, so included general info & snippets - hopefully that's all that's needed. Finally, yes - I'd personally like the option to not have anything extra displayed within TouchUI.   

If you need more info, please let me know...

**General Info**
- Ender 3 Pro
- OctoPrint 1.3.12 (192.168.0.75)
- Cura 4.4.1 - Windows 10 (192.168.0.25)

**Log Snippet 1**
- Test from a few minutes ago
- Deleted log then restarted OctoPrint
- Connected printer in WebUI ~14:15:40
- Attempted to start print from Cura @ 14:16:50
- Cura went to Monitor tab and would usually start pre-heating, BLTouch eval, then print, but nothing occurred except gcode uploaded to OP 
[Snippet1.log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/4145245/Snippet1.log)

**Log Snippet 2**
- Log snippet from 1/20 well before it stopped working
- Snippet starts at "Starting analysis of local" line of a print as that's line Cura comm seemed to start  in prior log & there was nothing substantial for the few minutes prior to that point
- Cut snippet once it seems printing had started and was working
[Snippet2.log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/4145246/Snippet2.log)

Thanks!

BzowK, 2020-02-03:
UPDATE:

I disabled the plugin and tested again, but issue remains so may be due to something else.  Investigating and will let you know.  

Thank You

OllisGit, 2020-02-03:
Hi @BzowK,
I don't use the autostart feature from cura, but I could reproduce the error.
It is related to the "Ultimaker Format Package" Plugin. Deactivate it and autostart is working.

I have an idea why this is happened:
My assumption is that the Cura-Plugin detects that it is possible to send a *.ufp file, instead of *.gcode. The ufp file is just a zip file and includes the gcode and some other meta-information like the thumbnail. 
So, the ufp file is send to octoprint with one api-call: "upload and print", but after uploading ocotprint can't select the ufp-file directly, because the gcode is not totally unpacked and stored in the upload folder.
I will look into this a little bit deeper, maybe with support from the other plugin developers (cura and ufp)
BR
Olli

OllisGit, 2020-02-03:
Hi @BzowK 
I raise two tickets maybe the other guys could find the issue a little bit faster:

https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/146
https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/issues/8

fieldOfView, 2020-02-03:
@BzowK what version of the OctoPrint Connection plugin in Cura do you use? The current one from the Marketplace?

OllisGit, 2020-02-03:
Hi @BzowK ,
I installed the CURA Octoprint Version 3.5 from the homepage: 
https://github.com/fieldOfView/Cura-OctoPrintPlugin

With this version the autoprint-feature works for me.

BzowK, 2020-02-12:
Thanks for the replies & sorry for the delay...

Great that you found the issue.  Poor troubleshooting by me, but had disabled related plugins except that one thinking it couldn't be the cause :(.  By the way, yes - I'm using the latest build from the Marketplace: 3.5.11.  

To note, I uninstalled the Marketplace version and replaced it with the Github (manually installed) version and its automatically printing now again, too.  Really odd.

Thanks, Guys!

fieldOfView, 2020-02-13:
An updated version of the Cura OctoPrint Connection plugin should be available from the Marketplace soon.

OllisGit, 2020-02-13:
@fieldOfView thanks a lot for your support!!

closed 2020-02-13 by OllisGit