OllisGit/OctoPrint-PrintJobHistory #20: Option to ignore failed prints

steveheinsch, 2020-02-11:
It would be really nice to have an option to auto delete failed prints.

I was trying to print a model yesterday and get it dialed in. So I was canceling prints often to make a tweak and start over. When you cancel a print, you want to raise the print head asap so it doesn't start melting the plastic where it froze in place after canceling, or even melt the PEI. It's hard to do that when you get 1) PJH popup that print failed (edit) and 2) Are you Sure? when you try to delete the failed job. Once you get past those 2 prompts, you can raise the print head. It's annoying to have to fight those prompts to begin with in order to be able to gain control of the printer by going to the control tab.

I'd like the opportunity to edit a job when successful, and be prompted for that, but delete it if not and not get these prompts.

OllisGit, 2020-02-12:
Hi @steveheinsch,
deleting (or move) a print job is possible with my other plugin: 
https://github.com/OllisGit/OctoPrint-DeleteAfterPrint
(it is also listed in the offical repo)

The second point: totally agree, it is on my list!
But first I want to finished my implementation with my current feature setup and bring that version to the offical repository.

OllisGit, 2020-05-03:
Hi,
it's fixed in the new release candidate, see release description: https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.0.0rc6

closed 2020-05-03 by OllisGit