OllisGit/OctoPrint-PrintJobHistory #21: [Enhancement] Ability to reprint directly from the history tab

BrandweerAA, 2020-02-22:
It would be very convenient to be able to store the actual gcode file so you can quickly re-print a file from history. Especially because, if you send the file directly to Octoprint from your slicer, no real source file is stored on your (slicing) device and requires an extra step to save to a project or something. And for users concerned about SD storage, you  could make it configurable, users can decide about using the feature themselves.

A possible solution could be to have a third button in the history list that transfers the selected file to the 'files' section on the main screen.

OllisGit, 2020-02-23:
Hi @BrandweerAA ,
nice idea and it is already on my long-term list (https://github.com/OllisGit/OctoPrint-PrintJobHistory/projects/1#card-27068042).

At the moment I'm not sure if I should store just a file-reference of the gcode (path to upload-folder/sd-path) or should I store the gcode into my database or should I copy the gcode to a separate history-folder.

Of course all options would be nice (selectable by user), but I need to start with something .

BrandweerAA, 2020-02-23:
Hi Ollis,

Oh, didn't see that one.

I would opt for one of the last two you mentioned. Can't speak for anyone else but my workflow is that after printing, I delete the file from the main page list because I like that list short. 

As for the db or the file system; I always try to use the file system myself for such things because you can access that in multiple ways (such as ssh) and you don't have to code specific db retrieval stuff to recreate the file. So if you ask me, I'd go for a separate history folder ;-)

closed 2020-09-09 by stale[bot]

OllisGit, 2021-05-24:
It's in the latest release V1.14.0