OllisGit/OctoPrint-PrintJobHistory #23: History quit updating after the OctoPrint 1.4.0 update.

sgano, 2020-03-13:
Not sure if the issue is with this plugin or the latest update to OctoPrint.  But ever since upgrading to OctoPrint v1.4.0 this plugin no longer records and lists new prints.  Any ideas about what may be causing this? (I am using v1.0.0rc5)

OllisGit, 2020-03-14:
Hi @sgano ,

the RC work currently only with version 1.3.x.
The next version will be compatible with 1.4.x.
Please be patient.

BR
Olli

sgano, 2020-03-15:
No problem, thanks for creating and updating this plugin!  I was just trying to narrow down the issue (which could have been on my side) or at the very least be sure the issue was reported/documented.  Take care!

Kyred-Saken, 2020-03-18:
Same thing happened to me after updating OctoPrint a few days ago.

jbjones27, 2020-04-04:
Same here.  Are there any workarounds to get this working in 1.4.0 in the meantime?
Thanks for your plugin and your support!

alsand61, 2020-04-11:
Same here after updating to OctoPrint 1.4.0
Thanks for the great work, Olli.

Here's the log, in case it helps...

2020-04-11 10:59:37,215 - octoprint.plugin - ERROR - Error while calling plugin PrintJobHistory
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 328, in on_event
    self._createPrintJobModel(payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 245, in _createPrintJobModel
    tempBed = currentTemps[0]["bed"]["target"]
KeyError: 0

BzowK, 2020-04-21:
Ditto

joshlarsen48, 2020-04-23:
Same here. Thanks for your work Olli

Magikti, 2020-04-24:
just a workaround to unlock those waiting for the release: edit the file /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py
and replace:self._figtle_manager.get_metadata(payload["origin"], payload["file"])
by:
self._figtle_manager.get_metadata(payload["origin"], payload["name"])

Magikti, 2020-04-24:
and, TKS TKS TKS for this very good plugin!!! :)

alsand61, 2020-04-26:
@Magikti Thanks for the workaround !!

> just a workaround to unlock those waiting for the release: edit the file /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py
> and replace:self._file_manager.get_metadata(payload["origin"], payload["file"])
> by:
> self._file_manager.get_metadata(payload["origin"], payload["name"])

anhtuan445, 2020-04-28:
I'm not sure it is the correct fix:
Look at the documentation: [(https://docs.octoprint.org/en/master/events/)]

You see in **MetadataAnalysisFinished** (and othern events)
Deprecated since version 1.3.0:
**file**: the file’s path within its storage location. To be removed in 1.4.0.

But for this place we need the path and you are replacing it with  name. You shoud replace it with path so 

self._file_manager.get_metadata(payload["origin"], payload["**path**"])

Name might work as it also exists but the value in there is not the one we want (unless name is not just file name but also path+name?)

At least it seems so to me, correct me if I'm wrong :)

jbjones27, 2020-04-28:
Hi All,
  For me, the original replacement with “name” works fine:

       self._file_manager.get_metadata(payload["origin"], payload["name”])

@anhtuan445:  Did you test with “path”?  Did that work also?

Best regards,
  Jeff

rongith, 2020-04-29:
For me it is `self._file_manager.get_metadata(payload["origin"], payload["path"])` which is working. "name" doesn't work.
OctoPrint 1.4.0 running on OctoPi 0.17.0

anhtuan445, 2020-04-29:
For me path and name both worked and if I debugged the values then they both gave the file name. My files are on local storage (not sd card, so payload origin is local). But I have seen that for some the name works and for some the path.

You can check with self._logger.info(payload["name"]); and self._logger.info(payload["path"]); command and then look in octoprint.log file (from /home/pi/.octoprint/logs/ o from preferences tab and Logging). 

As this command needs the actual file path then path seems more appropriate, but how it works with name eludes me :D Hopefully someone with more knowledge will explain it.

My OctoPi is 0.16.0, OctoPrint 1.4.0 by the way.

OllisGit, 2020-05-03:
Hi,
it's fixed in the new release candidate, see release description: https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.0.0rc6

closed 2020-05-03 by OllisGit