OllisGit/OctoPrint-PrintJobHistory #25: Feature Request - Picture Delay Timer(s)

r1aven, 2020-04-10:
Add delay timer option, in seconds, prior to capturing picture of completed print.  This would allow for any post print g-code instructions to complete prior to capturing image.  In my case, this would allow the x-carriage to be removed from the frame of view prior taking the snapshot.

OllisGit, 2020-04-11:
Hi @r1aven ,
the snapshot is taken when OctoPrint send the "PRINT_DONE" event. 
Maybe the event is send before all gcode buffered commands where executed in the printer.

Currently (since end of last year) I want to finish my plugin for "ready-to-ship". 
I am working on basic features.....if I don't fix issues in my other plugins. 

So, adding new stuff is currently not possible.

Did you try to add a gcode delay to your files?
```
G4 P60000 will wait for one minute
```
Maybe that helps, see https://reprap.org/wiki/G-code#G4:_Dwell.

C ya,
Olli

r1aven, 2020-04-12:
Hello @OllisGit ,
That is perfectly understandable and I had seen where you were trying to get this into the official octoprint plugin repo.  However, you are absolutely correct and I don't know why I didn't think to add a dwell to my post print g-code instructions.  Thank you for the suggestion and the link to the wiki.  Cheers!

closed 2020-04-12 by OllisGit