OllisGit/OctoPrint-PrintJobHistory #27: get_metadata(payload["origin"], payload["file"])

OllisGit, 2020-04-18:
see https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/99
```
2020-04-18 08:22:09,327 - octoprint.plugin - ERROR - Error while calling plugin PrintJobHistory
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 340, in on_event
    self._printJobFinished("canceled", payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 277, in _printJobFinished
    self._createAndAssignFilamentModel(self._currentPrintJobModel, payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 155, in _createAndAssignFilamentModel
    fileData = self._file_manager.get_metadata(payload["origin"], payload["file"])
KeyError: 'file'
```

Magikti, 2020-04-24:
Hello, just to unlock those waiting for the release: edit the file :
/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py
and replace:self._figtle_manager.get_metadata(payload["origin"], payload["file"])
by:
self._figtle_manager.get_metadata(payload["origin"], payload["name"])

OllisGit, 2020-05-03:
Hi,
it's fixed in the new release candidate, see release descripton: https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.0.0rc6

closed 2020-05-03 by OllisGit