OllisGit/OctoPrint-PrintJobHistory #28: Add PrusaSlicerThumbnails

OllisGit, 2020-04-25:
https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails

jneilliii, 2020-05-05:
If utilizing the metadata `thumbnail` property then whatever changes you do to make this work will also allow the UltimakerFormatPackage plugin to also be compatible.

ikopytko, 2020-06-02:
I made an attempt to resolve this issue using @jneilliii propose to use metadata to find out thumb path, and it is kinda better than before but still contains some manual path-related operations because what other plugins are storing in metadata is path for tornado url handler, not path on the disk. 

That's mean if there exists any other plugin for thumbnails which is based on Ultimaker Format Package or PrusaSlicer Thumbnails thaen it will work without any futher modification.

jneilliii, 2020-06-03:
You could also just check against the `thumbnail_src` attribute to determine which plugin data folder the image is stored in. Each of my plugins use that to identify which plugin the thumbnail was generated from.

OllisGit, 2020-06-03:
hi,
the first version of ufp doesn't include the meata-data, that's the reason why it is based on absolut paths.

Thx, a lot for your pull request I appreciate it very much.
I will take a look in the next days.

jneilliii, 2020-06-03:
I think it did include it but it was a different path, ie a sub attribute of refs.

OllisGit, 2020-06-06:
Implemented since V1.1.0

closed 2020-06-06 by OllisGit