OllisGit/OctoPrint-PrintJobHistory #31: Don't call optional plug-ins "Missing" Plug-ins

JohnOCFII, 2020-05-14:
I installed and tested PrintJob History.  I may, or may not install your other plug-ins in the future.  The existing PrintJobHistory works well, but OctoPrint warns me very, very often that I'm missing plug-ins!  This is annoying.  Don't use this feature as a marketing billboard for your other plug-ins. 

Thank you.

[image]

OllisGit, 2020-05-20:
hmmm...just hit the checkbox "disable..." then you will never see this popup! 
Just read the content in the dialog, everytime it appears ;-)

Or does this checkbox don't work?

Btw. only "DisplayLayerProgress" is developed be me. The other plugins were developed by other Developers. So, it has absolutely nothing to do with a "marketing billboard"!!!

JohnOCFII, 2020-05-20:
Thank you for your reply.

Yes, if I select the "disable Plugin-Dependency check" I do stop receiving the message.  My initial reluctance to that was that I did not know of that message was specific to your plug-in, or if it set that flag across ALL OctoPrint plug-ins.

Thanks,

John

OllisGit, 2020-05-27:
Changed to "optional" in newest release 1.0.0.

closed 2020-05-27 by OllisGit