OllisGit/OctoPrint-PrintJobHistory #32: Is the Import function disabled on purpose?

FixxCZ, 2020-05-14:
Export works fine, but import has grayed out options and Browse button doesn't work.
[image]

OllisGit, 2020-05-20:
Hi @FixxCZ ,

because the csv-importer is not 100% tested I disabled this feature in the current release candidate.

This is the last function before I will but this plugin to version 1.0.0 and propagate to the offical Plugin-Repository.

FixxCZ, 2020-05-20:
OK, perfect.

OllisGit, 2020-05-27:
Hi @FixxCZ ,
release version 1.0.0 is out and includes the CSV-Importer.

Please test and give feedback!

Thx, in advance
Olli

FixxCZ, 2020-05-28:
@OllisGit I have tried to append this file, but it just stays on the CSV-Import-dialog forever. I suspect that issue could be in the last column which has the currency which should be "Kč" but was exported as "KÄŤ"
Is the database in UTF-8 ?

[OctoprintPrintJobHistory28-05-2020-append.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/4695050/OctoprintPrintJobHistory28-05-2020-append.zip)

OllisGit, 2020-06-06:
Hi @FixxCZ,
the new version 1.3.0 fixes multiple issues.
- Not the Database had issues it was the socket.js framework. It could not handle an error message that includes this ```0.00Kč```
- Reading the cost and the costunit was not working
- Your sample csv-file includes a lot of empty (includes only commas) lines. This was not handled in the right way

Please test and give feedback.

closed 2020-07-16 by stale[bot]