OllisGit/OctoPrint-PrintJobHistory #33: [Enhancement] Option to make the history read only

FixxCZ, 2020-05-14:
Hi, I'm running the printer in the small community environment where users have dedicated user login to Octoprint. I would like to use this plugin to measure each user usage of filament, but problem is that any user can delete the job history items to clear their activity. I assume it would be difficult to introduce a permission into the official permission groups to make it read only for only certain group of users. So good enough solution would be to make it possible to disable the job delete button in the settings for everyone, even for admins. If admin would need to remove some job, she/he can be enable temporarily in the settings.

OllisGit, 2020-05-20:
Hi @FixxCZ ,

I think this would be a very good feature! I added this feature to my planing-board: 
https://github.com/OllisGit/OctoPrint-PrintJobHistory/projects/1

To be honest, I never looked into the permission-concept of OctoPrint so I have no idea if its possible...should be. But I can also ask in the forum about some "implementation-snippets".

OllisGit, 2020-05-22:
@FixxCZ fyi: it is possible to add custom plugin-specific permissions, BUT only since OP 1.4.......but this is a good excuse to only support OP 1.4 in the future, no backward compatibility any more ;-)
https://docs.octoprint.org/en/master/plugins/hooks.html?highlight=permission#additional_permissions_hook

FixxCZ, 2020-05-22:
Sounds like plan :)  I currently resolved it by modifying the package and removing the buttons from the layout and it works for purpose I need it.

OllisGit, 2022-01-16:
Hi @FixxCZ,

it took a while....but in the next release I added two new permissions: Edit and Delete-PrintJob.
[image]
Also in the tab  list the delete and add functionality is gone.

The default groups "admin" and  "operator" will include theses permissions as a defualt, so you need to create your own group... or remove it from the operator.

FixxCZ, 2022-01-17:
@OllisGit  Thanks a lot, you are the rockstar! It will help me a lot.

OllisGit, 2022-01-22:
Hi @FixxCZ ,

new release is out and permission check is implemented.

Please test and give feedback.

Thx, in advance
Olli

closed 2022-03-04 by github-actions[bot]