OllisGit/OctoPrint-PrintJobHistory #35: can't install on octoprint with Python3.7

marekorok, 2020-05-19:
Hi,
I have Octoprint 1.4.0 and moved while ago to python 3, version 3.7.3
Trying to install this plugin but no luck, seems problem around sqlite3 and database creation.
Db indeed is created but 0 bytes in size.

```
pi@octopi:~/.octoprint/data/PrintJobHistory $ ls -l
-rw-r--r-- 1 pi pi 0 May 19 20:28 printJobHistory.db
```

Any clues?
marekorok

Here is the log I found:

```
2020-05-19 19:42:17,732 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python3 -m pip" as command to invoke pip
2020-05-19 19:42:18,242 - octoprint.util.pip - INFO - Version of pip is 20.0.2
2020-05-19 19:42:18,244 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python3.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2020-05-19 19:42:18,244 - octoprint.util.pip - INFO - ==> pip ok -> yes
2020-05-19 19:42:18,327 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2020-05-19 19:42:18,332 - octoprint.plugins.PrintJobHistory - INFO - Start initializing
2020-05-19 19:42:18,332 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Init DatabaseManager
2020-05-19 19:42:18,332 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Creating database in: /home/pi/.octoprint/data/PrintJobHistory/printJobHistory.db
2020-05-19 19:42:18,333 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Check if database-scheme upgrade needed.
2020-05-19 19:42:18,335 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('SELECT "t1"."databaseId", "t1"."created", "t1"."key", "t1"."value" FROM "pjh_pluginmetadatamodel" AS "t1" WHERE ("t1"."key" = ?) LIMIT ? OFFSET ?', ['databaseSchemeVersion', 1, 0])
2020-05-19 19:42:18,337 - octoprint.plugin.core - ERROR - Exception while initializing plugin PrintJobHistory, disabling it
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 3099, in execute_sql
    cursor.execute(sql, params or ())
sqlite3.OperationalError: no such table: pjh_pluginmetadatamodel
```

OllisGit, 2020-05-20:
short analyse: python 3 issue.

marekorok, 2020-05-20:
yeap, worked perfectly :-)
then waiting for release....

```
2020-05-20 16:35:15,649 - octoprint.plugins.PrintJobHistory - INFO - Start initializing
2020-05-20 16:35:15,649 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Init DatabaseManager
2020-05-20 16:35:15,649 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Creating database in: /home/pi/.octoprint/data/PrintJobHistory/printJobHistory.db
2020-05-20 16:35:15,650 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Check if database-scheme upgrade needed.
2020-05-20 16:35:15,652 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('SELECT "t1"."databaseId", "t1"."created", "t1"."key", "t1"."value" FROM "pjh_pluginmetadatamodel" AS "t1" WHERE ("t1"."key" = ?) LIMIT ? OFFSET ?', ['databaseSchemeVersion', 1, 0])
2020-05-20 16:35:15,654 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Create database-table, because didn't exists
2020-05-20 16:35:15,655 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('DROP TABLE IF EXISTS "pjh_temperaturemodel"', [])
2020-05-20 16:35:15,655 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('DROP TABLE IF EXISTS "pjh_pluginmetadatamodel"', [])
2020-05-20 16:35:15,656 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('DROP TABLE IF EXISTS "pjh_filamentmodel"', [])
2020-05-20 16:35:15,657 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('DROP TABLE IF EXISTS "pjh_printjobmodel"', [])
2020-05-20 16:35:15,659 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('CREATE TABLE IF NOT EXISTS "pjh_printjobmodel" ("databaseId" INTEGER NOT NULL PRIMARY KEY, "created" DATETIME NOT NULL, "userName" VARCHAR(255), "fileOrigin" VARCHAR(255), "fileName" VARCHAR(255), "filePathName" VARCHAR(255), "fileSize" INTEGER, "printStartDateTime" DATETIME, "printEndDateTime" DATETIME, "duration" INTEGER, "printStatusResult" VARCHAR(255), "noteText" VARCHAR(255), "noteDeltaFormat" VARCHAR(255), "noteHtml" VARCHAR(255), "printedLayers" VARCHAR(255), "printedHeight" VARCHAR(255))', [])
2020-05-20 16:35:15,681 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('CREATE TABLE IF NOT EXISTS "pjh_filamentmodel" ("databaseId" INTEGER NOT NULL PRIMARY KEY, "created" DATETIME NOT NULL, "printJob_id" INTEGER NOT NULL, "profileVendor" VARCHAR(255), "diameter" REAL, "density" REAL, "material" VARCHAR(255), "spoolName" VARCHAR(255), "spoolCost" REAL, "spoolCostUnit" VARCHAR(255), "spoolWeight" REAL, "usedLength" REAL, "calculatedLength" REAL, "usedWeight" REAL, "usedCost" REAL, FOREIGN KEY ("printJob_id") REFERENCES "pjh_printjobmodel" ("databaseId") ON DELETE CASCADE)', [])
2020-05-20 16:35:15,694 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('CREATE INDEX IF NOT EXISTS "filamentmodel_printJob_id" ON "pjh_filamentmodel" ("printJob_id")', [])
2020-05-20 16:35:15,706 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('CREATE TABLE IF NOT EXISTS "pjh_pluginmetadatamodel" ("databaseId" INTEGER NOT NULL PRIMARY KEY, "created" DATETIME NOT NULL, "key" VARCHAR(255) NOT NULL, "value" VARCHAR(255) NOT NULL)', [])
2020-05-20 16:35:15,719 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('CREATE TABLE IF NOT EXISTS "pjh_temperaturemodel" ("databaseId" INTEGER NOT NULL PRIMARY KEY, "created" DATETIME NOT NULL, "printJob_id" INTEGER NOT NULL, "sensorName" VARCHAR(255) NOT NULL, "sensorValue" VARCHAR(255) NOT NULL, FOREIGN KEY ("printJob_id") REFERENCES "pjh_printjobmodel" ("databaseId") ON DELETE CASCADE)', [])
2020-05-20 16:35:15,731 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('CREATE INDEX IF NOT EXISTS "temperaturemodel_printJob_id" ON "pjh_temperaturemodel" ("printJob_id")', [])
2020-05-20 16:35:15,743 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('INSERT INTO "pjh_pluginmetadatamodel" ("created", "key", "value") VALUES (?, ?, ?)', [datetime.datetime(2020, 5, 20, 16, 35, 15, 742783), 'databaseSchemeVersion', '2'])
2020-05-20 16:35:15,753 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Database tables created
2020-05-20 16:35:15,754 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Done DatabaseManager
2020-05-20 16:35:15,754 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Init CameraManager
2020-05-20 16:35:15,755 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Snapshot-Folderr:/home/pi/.octoprint/data/PrintJobHistory/snapshots
2020-05-20 16:35:15,755 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Done CameraMenager
2020-05-20 16:35:15,756 - octoprint.plugins.PrintJobHistory - INFO - Done initializing
```

OllisGit, 2020-05-27:
Version 1.0.0 is out and solves this issue!

marekorok, 2020-05-31:
working perfectly.
Thank you!

closed 2020-05-31 by marekorok