OllisGit/OctoPrint-PrintJobHistory #36: Always ask to update to last version

Schawy, 2020-05-26:
Hi,

I have installed the last version 1.0.0rc6.1 but i get always an popup to ask me to update to the V1.0.0rc6.

[image](https://drive.google.com/file/d/18OHBjYteRnrKDjcQkx80E_wFzPlgAi6v/view?usp=sharing)

Is it possible to check that ?

Thanks.

OllisGit, 2020-05-26:
Hi @Schawy,

I can't see the image: "Access Denied", but you can just Copy&Paste the image into a comment textbox. Add a comment, put the image into the clipboard and paste the image into the comment box.

Regarding your mentioned issue, it looks like a OctoPrint-Bug..or missing feature.
I ask the community about the version-detection, see https://community.octoprint.org/t/ordering-of-version-numbering/20313

Schawy, 2020-05-26:
[image]
Hi @OllisGit know the picture 😉

OllisGit, 2020-05-27:
Hi @Schawy ,
a technical description of the issue could be found in the above forum-link.

But I release Version 1.0.0. So, this issue is solved anyway.

closed 2020-05-28 by OllisGit