OllisGit/OctoPrint-PrintJobHistory #37: Error after import of old print jobs

taker218, 2020-05-28:
I just installed the current version and imported some of my old prints from the PrintHistory plugin.

It seemed to work and it shows me that there are now 149 prints (I imported 4 prints). But when I go to the last page, it doesn't reload the list.

In the octoprint.log I get the following error:
`2020-05-28 11:34:03,837 - tornado.access - ERROR - 500 GET /plugin/PrintJobHistory/loadPrintJobHistoryByQuery?from=140&to=10&sortColumn=printStartDateTime&sortOrder=desc&filterName=all (192.168.178.23) 194.36ms
`
This is the csv I imported (had to rename it to txt for github).
[import_test3.txt](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/4694647/import_test3.txt)
There are just two prints in there, since tried it with different prints, but the outcome was the same.
I also enabled the SQL debugging, but it seems like there is no output in the octoprint.log

I also downloaded the database file and had a look at it. I exported the 4 prints and attached them here as well.
[database-content.txt](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/4694675/database-content.txt)

Any idea why it doesn't want to reload the list?

OllisGit, 2020-05-28:
Thanks for reporting the issue!

I found the issue: In a past I defined the field "End Datetime" as mandatory. I changed it, but the ui-table expected this.
Now fixed in latest Version 1.0.0.

Also in the "import_test3" file is a typo. Currently every column name must match 1:1, this includes lower/upper-case.
Please change to ```Used Weight [g]``` 

Please test and give feedback.
Olli

taker218, 2020-05-29:
Seems to be working now. I imported the file but didn't have the latest version installed (didn't gave me an update notification, so I installed it manually with the URL).

Had the same problem as before, after the update to the latest Version everthing seems to work now.

So everything is working as it should now from my point.

taker218, 2020-05-29:
Just FYI:
I just imported all 679 old prints and it worked as expected.

Will try it on my other printer as well, but I think it will be the same there.

closed 2020-05-30 by OllisGit