OllisGit/OctoPrint-PrintJobHistory #39: Importing CSV file from PrintHistory

calania, 2020-05-29:
Hello, I am having problems importing my old print jobs. I used to use the old PrintHistory plugin [https://github.com/imrahil/OctoPrint-PrintHistory/issues](url) and exported my CSV file from that plugin. However, looking at the sample CSV file you provided it seems that the old PrintHistory exports the start time in different formats and has slightly different names for everything resulting in me getting an error when importing saying

            
Mandatory column is missing! 
'DurationFile NamePrint result [success canceled failed]Start Datetime [dd.mm.yyyy hh:mm]'
 
 Nothing to import!

Is there any chance that you maybe could add support for importing old PrintHistory CSV files? 

Here is part of my CSV file [https://pastebin.com/raw/B9063DPJ](url)

taker218, 2020-05-29:
I had the same problem and used a lot of "Excel magic" to convert the old csv to the new one.

But it would be super nice to have an easier option to import the old data to the new one.
I saw that the old plugin and the export was pretty rough.

Just an example:
For the timestamp I first had to divide it by 10 so I could get the "real timestamp" and then I could convert it to a normal date/time. I don't really know why this is the case, but yeah it was pretty weird :D

OllisGit, 2020-05-29:
I thought creating a CSV-File (from any source) is an easy task. As mentioned, (only) some "Excel-Magic" is needed
 
That is also the reason why the csv-importer doesn't stick on a fix column order.
So, you can just grab the sample-file as a starting point and add full columns from the PrintHistory-Export just next to the sample column.

Or maybe I can create an Excel-File with two worksheets. 
In the first sheet you can copy the your PrintHistory-Data and the second worksheet includes the new headers and "reference-formulars" to grab the correct values/formats from your data.

mkuf, 2020-05-29:
I created a bash script that dumps PrintHistories' Database into importable CSVs
It might be helpful to some of you, have a look: https://gist.github.com/mkuf/5afd96404e980e7f0988103da934cf46

OllisGit, 2020-06-01:
Hi @mkuf ,
thanks a lot for the script...didn't test it, but looks good ;-)

I added a section in my wiki-page: https://github.com/OllisGit/OctoPrint-PrintJobHistory/wiki/Troubleshooting#import-of-printhistory-data

calania, 2020-06-02:
Hello, sorry for not answering sooner. I tried the script linked above and it worked! The only thing that wasn't clear was that you were supposed to use a db file stored on the pi and not the CSV file exported from the plugin. But once I figured that out everything worked!

OllisGit, 2020-06-02:
Hi, 
I reopened the issue, because in the next release there will be an "PrintHistory" CSV-Exporter.

That exporter use also the database, because the CSV created from UI is not complete.
The CSV-File could be "cleaned" by you (Excel or Texteditor) and after that imported to PJH.

OllisGit, 2020-06-06:
Hi,
in the lasted version 1.3.0 includes the exporter.

Please test and give feedback.

TheRealAvise, 2020-07-14:
Hi,

Just to let you know, I have an extensive PrintHistory database, which contains around 900 entries. Export function works great, a file gets generated and it looks OK, but when trying to import that data it would always hang. 

Looking at the script done by @mkuf I noticed that he mentions that import would hang if the file contains more than a certain number of entries.

Once I split the exported file into 3 files each containing 300 odd entries the import function worked great.

OllisGit, 2020-08-05:
Hi @TheRealAvise ,
thx for the feedback!

fyi I created an issue for the 900 line import issue.

closed 2020-08-05 by OllisGit