OllisGit/OctoPrint-PrintJobHistory #40: Error installing

mdietrich16, 2020-05-29:
This is probably not a problem with your plugin, so feel free to close this issue.
During the install process, octoprint tells me that building the wheel for pillow has failed.

I'm running this on Python version 2.7.16 in a virtual environment only for octoprint.
The complete stacktrace after collecting and downloading all dependencies (I edited those long "copying src/PIL..." lines out) is .

Should I install jpeg as mentioned in the stacktrace in the venv?

`Building wheel for pillow (setup.py): started
ERROR: Command errored out with exit status 1:
command: /home/pi/OctoPrint/venv/bin/python2 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-8syIy_/pillow/setup.py'"'"'; __file__='"'"'/tmp/pip-install-8syIy_/pillow/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-s_vyhz
cwd: /tmp/pip-install-8syIy_/pillow/
Complete output (141 lines):
running bdist_wheel
running build
running build_py
creating build
creating build/lib.linux-armv7l-2.7
creating build/lib.linux-armv7l-2.7/PIL
copying
...
copying src/PIL/FontFile.py -> build/lib.linux-armv7l-2.7/PIL
running egg_info
writing src/Pillow.egg-info/PKG-INFO
writing top-level names to src/Pillow.egg-info/top_level.txt
writing dependency_links to src/Pillow.egg-info/dependency_links.txt
reading manifest file 'src/Pillow.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching '*.c'
warning: no files found matching '*.h'
warning: no files found matching '*.sh'
warning: no previously-included files found matching '.appveyor.yml'
warning: no previously-included files found matching '.coveragerc'
warning: no previously-included files found matching '.codecov.yml'
warning: no previously-included files found matching '.editorconfig'
warning: no previously-included files found matching '.readthedocs.yml'
warning: no previously-included files found matching 'azure-pipelines.yml'
warning: no previously-included files matching '.git*' found anywhere in distribution
warning: no previously-included files matching '*.pyc' found anywhere in distribution
warning: no previously-included files matching '*.so' found anywhere in distribution
no previously-included directories found matching '.azure-pipelines'
no previously-included directories found matching '.travis'
writing manifest file 'src/Pillow.egg-info/SOURCES.txt'
running build_ext
The headers or library files could not be found for jpeg,
a required dependency when compiling Pillow from source.
Please see the install instructions at:
https://pillow.readthedocs.io/en/latest/installation.html
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-8syIy_/pillow/setup.py", line 903, in <module>
raise RequiredDependencyException(msg)
__main__.RequiredDependencyException:
The headers or library files could not be found for jpeg,
a required dependency when compiling Pillow from source.
Please see the install instructions at:
https://pillow.readthedocs.io/en/latest/installation.html
----------------------------------------
ERROR: Failed building wheel for pillow
Building wheel for pillow (setup.py): finished with status 'error'
Running setup.py clean for pillow
Building wheel for peewee (setup.py): started
Building wheel for peewee (setup.py): finished with status 'done'
Created wheel for peewee: filename=peewee-3.13.3-py2-none-any.whl size=127265 sha256=7c74ff9929b31fc6cb7aa9d38bde0763be8429b68eb59de5e956f04f2a4ec55b
Stored in directory: /tmp/pip-ephem-wheel-cache-GLe_kF/wheels/02/12/fa/72f23f801097e35a873dd9e6ba181a62c912f0a75dcc308548
Successfully built Print-Job-History peewee
Failed to build pillow
Installing collected packages: pillow, peewee, Print-Job-History
Running setup.py install for pillow: started
ERROR: Command errored out with exit status 1:
command: /home/pi/OctoPrint/venv/bin/python2 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-8syIy_/pillow/setup.py'"'"'; __file__='"'"'/tmp/pip-install-8syIy_/pillow/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-EdIRYu/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/OctoPrint/venv/include/site/python2.7/pillow
cwd: /tmp/pip-install-8syIy_/pillow/
Complete output (141 lines):
running install
running build
running build_py
creating build
creating build/lib.linux-armv7l-2.7
creating build/lib.linux-armv7l-2.7/PIL
copying src/PIL/PixarImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
...
copying src/PIL/FontFile.py -> build/lib.linux-armv7l-2.7/PIL
running egg_info
writing src/Pillow.egg-info/PKG-INFO
writing top-level names to src/Pillow.egg-info/top_level.txt
writing dependency_links to src/Pillow.egg-info/dependency_links.txt
reading manifest file 'src/Pillow.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching '*.c'
warning: no files found matching '*.h'
warning: no files found matching '*.sh'
warning: no previously-included files found matching '.appveyor.yml'
warning: no previously-included files found matching '.coveragerc'
warning: no previously-included files found matching '.codecov.yml'
warning: no previously-included files found matching '.editorconfig'
warning: no previously-included files found matching '.readthedocs.yml'
warning: no previously-included files found matching 'azure-pipelines.yml'
warning: no previously-included files matching '.git*' found anywhere in distribution
warning: no previously-included files matching '*.pyc' found anywhere in distribution
warning: no previously-included files matching '*.so' found anywhere in distribution
no previously-included directories found matching '.azure-pipelines'
no previously-included directories found matching '.travis'
writing manifest file 'src/Pillow.egg-info/SOURCES.txt'
running build_ext
The headers or library files could not be found for jpeg,
a required dependency when compiling Pillow from source.
Please see the install instructions at:
https://pillow.readthedocs.io/en/latest/installation.html
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-8syIy_/pillow/setup.py", line 903, in <module>
raise RequiredDependencyException(msg)
__main__.RequiredDependencyException:
The headers or library files could not be found for jpeg,
a required dependency when compiling Pillow from source.
Please see the install instructions at:
https://pillow.readthedocs.io/en/latest/installation.html
----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/OctoPrint/venv/bin/python2 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-8syIy_/pillow/setup.py'"'"'; __file__='"'"'/tmp/pip-install-8syIy_/pillow/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-EdIRYu/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/OctoPrint/venv/include/site/python2.7/pillow Check the logs for full command output.
Running setup.py install for pillow: finished with status 'error'
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output`

rtrainer, 2020-05-29:
I have the same issue

mkuf, 2020-05-29:
>Should I install jpeg as mentioned in the stacktrace in the venv?

You'll need the dev package of that lib. 
Also zlib is a requirement for pillow, so `apt install zlib1g-dev libjpeg8-dev` should do the trick.

OllisGit, 2020-05-29:
Hi @mkuf,

thx a lot for finding a solution!
 
I looked into other op-plugin issue-trackers and found similar-issues with pillow in OctoSlack and Octolaps.
Octolaps wrote an wiki-page about pillow-installation. 

If the mentioned command is not working take a look in this page: https://github.com/FormerLurker/Octolapse/wiki/V0.4---Troubleshooting-Installation-Issues#failed-to-build-pillow

Please give a feedback.

mdietrich16, 2020-05-31:
Yes, that fixed it. Very happy, thank you both!

closed 2020-05-31 by mdietrich16