OllisGit/OctoPrint-PrintJobHistory #41: Can't export CSV

taker218, 2020-05-29:
Me again,

I tried to export my prints as csv, but I just get an error 500 and the following in the octoprint.log:
```
2020-05-29 11:52:10,756 - tornado.application - ERROR - Uncaught exception GET /plugin/PrintJobHistory/exportPrintJobHistory/CSV (::ffff:192.168.178.23)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 1489, in _execute
    result = self.prepare()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 257, in prepare
    self._fallback(self.request, b"")
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 621, in __call__
    response.extend(app_response)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/werkzeug/wsgi.py", line 506, in __next__
    return self._next()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/werkzeug/wrappers/base_response.py", line 45, in _iter_encoded
    for item in iterable:
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/common/CSVExportImporter.py", line 401, in transform2CSV
    csvColumnValue = '"' + csvColumn.getCSV(job)  + '"'
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/common/CSVExportImporter.py", line 54, in getCSV
    columnValue = str(columnValue)
UnicodeEncodeError: 'ascii' codec can't encode character u'\xdf' in position 11: ordinal not in range(128)
2020-05-29 11:52:10,761 - tornado.access - ERROR - 500 GET /plugin/PrintJobHistory/exportPrintJobHistory/CSV (::ffff:192.168.178.23) 5876.74ms
```

I activated SQL debugging and there is now a lot more in the log. I attached it here.
[sql_debug_export_csv.txt](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/4701028/sql_debug_export_csv.txt)

OllisGit, 2020-05-29:
Hi @taker218 ,
you have a ```ß``` in your database or a "German-Umlaut". UUnfortunately there is a difference between python3 and 2.

Which version do you use?

taker218, 2020-05-29:
I think python 2, I have an older version of octopi running.

I'll have a look at the database when I'm home again.

taker218, 2020-06-01:
With the latest version of the plugin the export works like expected.

closed 2020-06-01 by OllisGit