OllisGit/OctoPrint-PrintJobHistory #43: [Feature Request] Selection Checkboxes

KenLucke, 2020-05-29:
Please add selection checkboxes for each item, which could be selected and use to determine a number of jobs to be deleted or exported at one time.

This would be comprised of a manual checkbox next to each job, and a "Select All/Select All This Page/Toggle Seleced" Drop-down menu at the top, which would be used by a "Delete Selected/Export Selected to CSV" option.

OllisGit, 2020-05-29:
Good idea! I will put that to my backlog.

OllisGit, 2020-06-03:
Btw. you can always clean the whole database via the plugin-settings menu!

OllisGit, 2020-09-19:
The row selection feature is available in the version 1.8. 
Please test and give feedback.

BR
Olli

KenLucke, 2020-09-19:
Will do.  I have it turned off at the moment, as it was getting tedious to delete all the jobs one at a time.  I'll turn it back on after current jobs are done and test.

closed 2020-10-29 by stale[bot]