OllisGit/OctoPrint-PrintJobHistory #44: Please clearify plugin dependencies.

AndKe, 2020-05-30:
A new user is greeted with word "dependencies" which is kind of the opposite of "optional":
----------
To get the real power of the Print Job History - Plugin you need the following plugins installed/enabled:

PreHeat (missing)
FilamentManager (missing)
DisplayLayerProgress (missing)
UltimakerFormatPackage (missing)
---------------
Please describe a little about what use they provide... what kind of statistics or what a user will be missing out by not having them. And, can any of those be added, producing extra data, at a later time.
For the majority of users, downloading UltimakerFormatPackage does not make sense as we don't use ultimaker.

Schokobecher, 2020-05-30:
FilamentManager is also not Python3 compatible since it's abandoned so it would make sense to ditch as many "dependencies" or "optional" things as possible.

OllisGit, 2020-05-31:
Hi,
I changed the text for more clarification in the newest release 1.1.0.

leoheck, 2020-12-20:
I'm still seeing this.
```PreHeat (missing)```

Which one should I install?
[image]

OllisGit, 2021-01-17:
Hi @leoheck ,

choose "Preheat Button".

I will update the Dialog in the next release to make it more clear.

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

leoheck, 2021-01-23:
@OllisGit No problem, thanks for the feedback and the help. This is something needed for new users like me.

OllisGit, 2021-04-11:
fyi: Currently I have some "private challenges to solve", thats the reason why It takes so long to release a new version, sorry!

Today I released a new development version (1.11.0). This version fix the issue. Please test and give feedback.

Thx, in advance
Olli

closed 2021-05-16 by OllisGit