OllisGit/OctoPrint-PrintJobHistory #47: WebCam Stream not loading within the Print Job Dialog

mkuf, 2020-05-30:
Hey there, 

I just installed this Plugin and I'm experiencing an issue with the Image Feature. 
When opening a logged Print, the capture button is blanked out, but is still klickable. 
[image]

Proceeding in wanting to capture an Image I'm greeted with a message that the Stream isn't loaded. 
[image]

At this point, the following Log appears in the octoprint.log
```
2020-05-30 17:09:22,492 - octoprint.server.api - ERROR - Error while running a test GET request on None
Traceback (most recent call last):
  File "/opt/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 486, in _test_url
    with requests.request(method=method, url=url, timeout=timeout, verify=valid_ssl, stream=True) as response:
  File "/opt/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/opt/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/requests/sessions.py", line 516, in request
    prep = self.prepare_request(req)
  File "/opt/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/requests/sessions.py", line 459, in prepare_request
    hooks=merge_hooks(request.hooks, self.hooks),
  File "/opt/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/requests/models.py", line 314, in prepare
    self.prepare_url(url, params)
  File "/opt/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/requests/models.py", line 388, in prepare_url
    raise MissingSchema(error)
MissingSchema: Invalid URL 'None': No schema supplied. Perhaps you meant http://None?
```

The stream itself is working fine within octoprint but it seems that the Plugin is unable to access it. 
[image]

Any Idea what might cause this issue?
-Markus

OllisGit, 2020-05-30:
Hi @mkuf ,
I only tested with an IP-Cam, because I didn't have a raspi-cam or a special streaming-cam.

Try to enter this url in your web-stream settings: ```http://<yourIP>/webcam/?action=stream```
First you can test the url in your browser, The browser should display the camera-stream.

Also I am reusing the original implementation from OP...maybe I missed something...will take a look.

mkuf, 2020-05-30:
Hi @OllisGit 
thanks for the reply. 

I just changed the URL as per your suggestion but unfortunately the issue still remains. 
The Stream is available via http://10.0.0.21:8080/?action=stream but it is not loading within the Plugin.
[image]
The logged error is identical to my previous attemts. 

-Markus

OllisGit, 2020-05-31:
Hi @mkuf ,

did you configure both URLs (stream and snapshot)?

I asked an other User with a Raspi-Cam and he is able to use the capture feature. His settings looks like this:

[image]

Please also open the Browser Developer-Tools to see if there are some errors in the Console:
- press F12 to open the dev-tools and switch to console tab
- reload the page
- look for red error messages

mkuf, 2020-05-31:
Oh, that makes sense 😅 
In my case the Snapshot URL was empty, as I don't use Octoprints Timelapse Feature. 
Setting it to a valid URL fixed the issue. 

[image]

Thanks!
-Markus

closed 2020-05-31 by mkuf