OllisGit/OctoPrint-PrintJobHistory #48: Install error in FreeBSD jail

maziggy, 2020-05-31:
Plugins doesn't install properly in FreeBSD jail. No error messages during install but it just don't get loaded due obvoiously missing sqlite Python module.

octoprint.log:

`2020-05-31 12:23:16,968 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin did not pass check, not loading.
2020-05-31 12:23:17,007 - octoprint.plugin.core - ERROR - Error loading plugin PrintJobHistory
Traceback (most recent call last):
  File "/data/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugin/core.py", line 967, in _import_plugin
    module = _load_module(module_name, spec)
  File "/data/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugin/core.py", line 71, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/data/OctoPrint-PrintJobHistory/octoprint_PrintJobHistory/__init__.py", line 26, in <module>
    from .DatabaseManager import DatabaseManager
  File "/data/OctoPrint-PrintJobHistory/octoprint_PrintJobHistory/DatabaseManager.py", line 8, in <module>
    import sqlite3
  File "/usr/local/lib/python2.7/sqlite3/__init__.py", line 24, in <module>
    from dbapi2 import *
  File "/usr/local/lib/python2.7/sqlite3/dbapi2.py", line 28, in <module>
    from _sqlite3 import *
ImportError: No module named _sqlite3
`

OllisGit, 2020-06-01:
hmm..thats strange because the sqlit3 is included since 2.5, see https://docs.python.org/2.7/whatsnew/2.5.html#the-sqlite3-package

But it looks like a common problem under some Unix-System, see https://stackoverflow.com/questions/1210664/no-module-named-sqlite3

and this looks promising:
```
sudo apt-get install libsqlite3-dev
```

Please give me a feedback, if you found a solution. Then I can add this to a "trouble shoting"-page.

maziggy, 2020-06-01:
Just checked...a simple 'pkg install py27-sqlite3-2.7.18_7' and everything is good ;)

closed 2020-06-01 by maziggy