OllisGit/OctoPrint-PrintJobHistory #49: Single Extruder Print on Dual Extruder System

jph290, 2020-06-01:
I am using FilamentManager with my octoprint and I have a dual extruder system.  My prints typically only use 1 extruder and my FilamentManager is setup with both extruders.  When I print using T1 (extruder 2) it brings in information based on extruder 1.  I understand tracking multiple extruders in a print would be an issue, but I would think there should be a way to determine which extruder was used to print and grab that data.  Currently this plugin gives me no use because it fills no information in to the usage and puts in the Tool 0 spool info.

jph290, 2020-06-01:
I can take a look when I have a chance and see if I can find a way to do this and make a pull request if you don't have time to look into this as well.  I think the plugin is great but this small tweak will make it much more useful.

OllisGit, 2020-06-02:
Hi @jph290 ,

currently only a single-extruder is supported. I think I added some todos in the sourcecode.
I will add this feature to my backlog (https://github.com/OllisGit/OctoPrint-PrintJobHistory/projects/1) and I will also add a note to this limitation to the readme.

Pull-Request were always welcome!!!!
Feel free to implement this feature.

Thx, for reporting this.
Olli

closed 2020-09-09 by stale[bot]