OllisGit/OctoPrint-PrintJobHistory #50: changes to settings recorded?

cranky-coder, 2020-06-01:
Periodically while printing, i may change the temperature, the speed and the extrusion rate through octoprint.  Any chance this plugin will detect those changes?  would be nice to get a final output on whatever the last settings were since i may tweak them having moved to a different filament

OllisGit, 2020-06-03:
Well...thats a total different story...almost...because you need something like a "timeseries-database" (see https://en.wikipedia.org/wiki/Time_series_database).

You want to store a huge amount of data and of course you want to query/filter the data in a performant way.

I have that on my [roadmap](https://github.com/OllisGit/OctoPrint-PrintJobHistory/projects/1) (long-term) and I also played around with different databases, see https://community.octoprint.org/t/prometheus-exporter/14047/15

So, that would be really nice feature for octoprint...but currently I don't have time to implement it....maybe later.

cranky-coder, 2020-06-03:
I didn't even think of a time series db.  Which is odd since i have a fairly large ELK setup here.  But that would make complete sense.  Something that can either be scraped (prometheus) or something to push (influxdb) would be perfect, all the current settings at that time could be recorded at that exact moment in the print.  Ill keep an eye out and maybe do some tinkering myself.  If i stumble across anything ill report back!

closed 2020-09-09 by stale[bot]