OllisGit/OctoPrint-PrintJobHistory #52: Nozzle Temp isn't showed in history

chimcen, 2020-06-03:
Hi,

thank you for this awesome plugin.
All is working great - excetp the nozzle temp is shown as 0°C (insted of 195°C).
I'm using latest octoprint with an Longer LK4 and Cura 4.6.1...

Any ideas?

[image]

OllisGit, 2020-06-03:
Hi @chimcen,

do you installed the [PreHeat](https://plugins.octoprint.org/plugins/preheat/)plugin?
Because it provides the initial Bed/Nozzel-Temperature, if not my plugin just ask the printer about the current temp after print starts with a 10 second delay...maybe after 10 secods the temp is still 0.
Or do you have multiple extruder, because the current implementation only picks the temp from tool0.

chimcen, 2020-06-03:
Yes, this is installed - now it mekes sense, because in PreHeat Plugin i set the temp for the bed to 45°C and for the nozzle to 0°C... because i is this only for heating up the bed for calibration releated things...

Is it possible for you plugin to get the temps from another source like the gcode file or similar...?

OllisGit, 2020-06-03:
Well, I don't want to parse the file, because other plugins did this already.

So, preheat-plugin parse the file and grabs all temp-informations, but I didn't realized that "somebody" turned the pre-heating of.

The layer-approach seems to be a good solution....but what should be the default solution...or even better the user can choose:

- If preheat is installed choose this value 
- If DisplayLayerProgress is installed, ask printer after Layer 3 about the current temperature
- If no plugin is installed, just grab the temp after x seconds (default)

...but this is maybe to complicated and confuse the user.

Or do you think an adjustable delay is totally enough (easier to implement/maintainable)? 
In your case, what delay would fit for you?

chimcen, 2020-06-03:
Yeah, you're right - include all three options inte the plugin is to much for some users.

I think this qould be the best option:

- If DisplayLayerProgress is installed, ask printer after Layer 3 about the current temperature

Make the "asking layer" editable from plugin settings would be the also great, but in the end go and ask at layer 3 should be enough...

OllisGit, 2020-06-06:
Hi @chimcen, 
in the newest Version it is now possible to adjust the delay in the plugin-settings.

The layer option was a little bit to complicated to implement...maybe in a later release...maybe not!

taker218, 2020-06-08:
Works for me, now both bed and nozzle temp are stored in the history.

Not using the preheat plugin, just adjusted the delay to my printer.

OllisGit, 2020-06-08:
Thx, for your feedback.

What delay value fits for you? Maybe I need to adjust the default value.

taker218, 2020-06-08:
I have set it to 120 seconds on my Ender 3 and 300 seconds on my CR10 (it needs a lot of time to heat the bed).
But I think the default value should be good.

Don't know, if the delay on my cr10 is enough since I have a 20h print running on it at the moment. Will see that tomorrow, when it's done.

chimcen, 2020-06-09:
Hi Olli,

thank you for this awesome work. As we talked about options - i think this is enough and also a good solution.
It works perfect.
If set mine to 600 sec - not because of heatup delays; it's more a matter about avoid problems when printing things with diff temps in first layers in addition to the heatup phase...
In my mind, the "final" print temp is interesting in the database entry.

Grüße aus Bayern und danke :)

taker218, 2020-06-09:
So the 5 minute delay on my CR10 wasn't quite enough, but I guess the default delay is more than good enough.
It really depends on how the user heats up his printer. Some might heat the bed first and then the nozzle (like I do) some heat it up at the same time, some preheat it (with or withou plugin). I guess there are a lot of combinations how to heat up your printer, so don't worry too much on adjusting the default delay.

btw: I just noticed a little typo in the settings.
It says "Temerature" instead of "Temperature"

closed 2020-06-21 by OllisGit

baschdl78, 2022-06-17:
Hello an before reporting my issue THX for the great work with your plugins... i love them!!!
I think it would be better to grab the nozzle temp on a specific layer...  otherwise, espacially on printers with larger volumes it could hapen that your still on the first layer (temp higher than used for the other layers) or your delay is way to long and the print is finished before the printer reads the temp.
It would be nice if the nozzle temp could be edit afterwards to "correct" wrong readings.
On my mainly used printer (Hypercube Evo 300x300x375  single extruder with Preheat, SpoolManager and PrintJobHistory Plugin installed) the nozzle temp is always reported as -1°C... i tried different settings but couldn't find a working solution.
Thank You, und viele Grüße aus Karlsruhe!

Sebastian