OllisGit/OctoPrint-PrintJobHistory #53: [Request] Support for M240 GCode for the final Image

Bootscreen, 2020-06-05:
Hey,

is it possible that you react to the M240 gcode in the "on job complete" scripts? This gives us the possibility that we can move the Object to a good position and create the screenshot without our impact.

i hope you understand what i mean.

OllisGit, 2020-06-06:
Hi @Bootscreen ,

I am not sure if I understand your request.
- PJH take the snapshot after all gcode commands were executed. This includes also the OP-Section "After print job completes".

Do you want to take a picture earlier? During execution of the script?

Please explain a little bit deeper.

Bootscreen, 2020-06-06:
Exactly, i want to make the Finish Screenshot during the "after job complete" gcode, because Printjob History makes the Screenshot at the end, this means on my ender 3 at y max, and y max is half out of my cam.

See this:
[image]

when i want a good photo i must push the bed back and make a screenshot manual

i hope you understand it now? when you speak german i can it explain it more detailed ^^

OllisGit, 2020-06-12:
Hi @Bootscreen ,

the new version 1.4.0 is now available and ready for testing.

Please give a feedback if it works for you.

Bootscreen, 2020-06-14:
It works =)

But one Info, maybe for the Wiki, when you use Cura you must put the GCode Command in the Printer End Code on Cura, because normaly there is the Code which moves the bed to the end point to present the print ^^

OllisGit, 2020-06-21:
Hi @Bootscreen!

Thx, for the feedback. I am not sure why the info about Cura is so important, because if someone is using the M240 code, he should know how to handle it.
Maybe I didn't get it.

closed 2020-09-09 by stale[bot]