OllisGit/OctoPrint-PrintJobHistory #54: [Behavior Change Requested]

KenLucke, 2020-06-05:
I often have more than one browser windows open for each instance of OP running... a Dashboard window and the Gcode viewer (as long as the file isn't too big) and/or the Control or Terminal Tab in another window(s).  When a print finishes, I get the dialog to save the history info in all open windows for that printer.  Can you add a check somehow to only allow that dialog to pop up in one window, or dismiss it or dim the save button in one if the other has already been saved?  

I realize that they are both (all) pulling the same browser data from the OP server.  Perhaps using a "dirty" flag that is checked every [x] amount of time to see if the info still needs to be saved?

OllisGit, 2020-06-06:
Hi @KenLucke ,
in the newest version 1.3.0 the Edit-Dialog is closed after you close/save it in one of your open browsers.

Please test and give feedback!

KenLucke, 2020-06-06:
> Hi @KenLucke ,
> in the newest version 1.3.0 the Edit-Dialog is closed after you close/save it in one of your open browsers.
> 
> Please test and give feedback!

Will do.  Starting a 2 day print now, so it will be after that ends.

Thanks!

KenLucke, 2020-06-08:
> Hi @KenLucke ,
> in the newest version 1.3.0 the Edit-Dialog is closed after you close/save it in one of your open browsers.
> 
> Please test and give feedback!

Was able to get printer #2 online, and test.  Works perfectly.  Had 4 browser windows open from the same print on purpose to test it.  Clicked "Save" in one, and the other three went away.  Perfect.

Thank you!

closed 2020-06-08 by OllisGit