OllisGit/OctoPrint-PrintJobHistory #57: Prusa Slicer Thumbnail is not used even though it is present in the GCode

NameOfTheDragon, 2020-06-09:
I'm not getting the thumbnails as expected in the print job history. I have the option selected to "_`Use thumbnail (if provided) after print, instead of snapshot`_" selected. The thumbnail is definitely present because it's showing up in the file list window (screen shot)

[image]

My settings are as follows:
[image]

I have attached the GCode file (downloaded from OctoPrint) that was used to print the item in the screen shots. You will see that it contains 2 thumbnails, 16x16 and 220x124

[EXTRUDER_Gear_0.2n_0.1mm_PETG_ENDER3_55m.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/4752124/EXTRUDER_Gear_0.2n_0.1mm_PETG_ENDER3_55m.zip)

NameOfTheDragon, 2020-06-09:
By the way, thanks for a great plugin. I really like the idea of keeping a log of printed items and hopefully later being able to view stats.

OllisGit, 2020-06-12:
Hi @NameOfTheDragon ,

the new release is out V1.4.0, BUT I only changed the text and not the default behaviour.

Because "you" prefer the thumbnail and other people prefer the real deal and want a picture from the cam instead of the slicer-thumbnail.
So, I need to find an "nice" UI-Design how to handle such configuration. 

Ideas are welcome (keep in mind, that in future you can grab several pictures)!!!

NameOfTheDragon, 2020-06-12:
In your screen shot for release 1.4.0, the text hasn't actually changed... I'm running a print right now but as soon as it finishes (~ 1 hour) I'll install the update and see if it has actually changed.

In my previous post (last paragraph) I suggested how to handle this:

> Perhaps what is needed is a toggle control that toggles between "prefer thumbnails" or "prefer camera"?

The toggle control would be disabled if only one source is enabled. As soon as 2 sources are enabled, then you get to choose which one has priority.

You could default it to "Prefer camera" so that it's a non-breaking change for existing users.

NameOfTheDragon, 2020-06-12:
I've updated to the latest release and yeah, the text hasn't changed. That's OK, the text describes the exact behavior I want ;-) What I want it to be able to prioritize thumbnails over snapshots.
[image]

P.S. You've labelled this issue as a "question" - it actually isn't. Depending on your perspective, it is either a bug or an enhancement request.

OllisGit, 2020-08-05:
Hi @NameOfTheDragon ,
in the latest version 1.6.0 I added a new feature to select the prefered image-source.

Please test and give me a feedback.

Thx, in advance
Olli

NameOfTheDragon, 2020-09-06:
I think we can probably close this now. Thanks for all your work on this.

closed 2020-09-06 by NameOfTheDragon

Doprintityourself, 2023-01-30:
I have installed 1.17 of Printjobhistory but i do not get any Thumbnails any more:
[image]
I have Slicer Thumbnails latest 1.0.1 installed and can see every Thumbnail in  the status section but not in Print Job History after printing
[image]

mikaeltulldahl, 2023-03-13:
Hello I can second that it broke recently. It seems related to which slicer I'm using. With Prusaslicer 2.6.0-alpha5+win64 I will only get a photo in the end. I picked up an old .3mf project and sliced with Prusaslicer 2.5.0+win64 and got a thumbnail in the print job history