OllisGit/OctoPrint-PrintJobHistory #58: Populate Spool... fields from PrusaSlicer gcode comments

6ffm70, 2020-06-14:
**Background**
I have been waiting for a plugin like this. Currently, we record our prints in a LibreOffice table, and it's a lot of typing for something your plugin may do automatically.

**Problem**
We don't want to use the filament manager plugin to maintain our currently 74 spools, but still want to see our chosen spool description in the print history.

**Possible solution**
We're using PrusaSlicer and the current release returns our chosen spool description under 
`; filament_settings_id = "chosen spool description"`
Would it be possible to have a checkbox or a setting that the field _Spool_, _Material_ and _Vendor_ are populated from the settings commentary in the PrusaSlicer gcode files?

_Spool_ = `; filament_settings_id`
_Material_ = `; filament_type`
_Vendor_ = `; filament_vendor`  // though at the moment it says `(Unknown)`, and I can see no means to populate that field in PrusaSlicer

OllisGit, 2020-06-21:
Hi @6ffm70 ,

sounds like a good idea!

It looks like that this feature (exporting material-info) is not a supported for Cura and only available for PrusaSlicer.

OllisGit, 2021-01-17:
this ticket is related to #104

Short: Grab field-values from gcode instead from other plugins or own calculation.

@6ffm70 maybe you have a suggestion how a UI could look like.

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

6ffm70, 2021-02-02:
Hello @OllisGit , thank you for getting back on this.

My proposal to get the filament name from PrusaSlicer into the PrintJobHistory entry is the following:

[image]
1.) Name as saved in P/S

[image]
2.) Name as transmitted via gcode

[image]
3.) Additional option in P/J/H settings

[image]
4.) Pre-populated form field when print job window pops up

6ffm70, 2021-02-02:
In another finding: I sometimes use multiple spools per print, which are manually changed during on print job.

In PrusaSlicer, it looks like this:
[image]

In the gcode, it looks like this:
[image]

Eight different types is an extreme example; however this might occur across the field from time to time. It might make sense knowing this beforehand.

OllisGit, 2022-01-17:
fyi: I am starting writing a kind of concept to implement such feature. I am looking for a simple and easy to use solution.
https://github.com/OllisGit/OctoPrint-PrintJobHistory/wiki/GCodeSpoolInfo

closed 2022-02-28 by github-actions[bot]