OllisGit/OctoPrint-PrintJobHistory #60: F/R: Print farm mode: sync history between OctoPrint instances

6ffm70, 2020-06-14:
**Background**
We have a couple of different printers and a couple of OctoPrint instances in the same VPN. The printers rotate between locations, though the OctoPrint instances don't. Once the printer changes on an instance, the _printer profile_ is changed accordingly.

**Problem**
We want to have all print jobs/history in at least in one source, as neither the printer who did it nor the location (the instance) matters too much.

**Possible Solution**
Have every entry in the history have fields to show _printer profile_ and _OctoPrint:Title_ or _Hostname_; and the option to sync those different PrintJobHistory databases to show all entries (possibly with an option to filter by user, printer, host).

OllisGit, 2020-06-21:
Hi @6ffm70 ,

my goal is to support a single external database-instance, see https://github.com/OllisGit/OctoPrint-PrintJobHistory/issues/45.

So, if you have multiple printers, you can see all history-jobs from all instances in a single instance UI.

Currently, I am writing a small concept about this solution (UI-Design, Database-Structure, Upgrade-Mechanism, Thumbnail-Support, ...)....so It could take a while.

6ffm70, 2020-07-28:
Great! That looks pretty much like it does the job.

closed 2020-09-14 by stale[bot]