OllisGit/OctoPrint-PrintJobHistory #61: Filament Field Blank After Job

jaredarm, 2020-06-16:
I'm printing on a dual extrusion machine (Flashforge Creator Pro) and using the left extruder which is labelled as Tool 1 (the right extruder is Tool 0). The estimates for filament usage and height during a print are all shown in the State section but after a print is finished the fields on the job history related to the filament are all blank.

OllisGit, 2020-06-21:
Hi @jaredarm ,

in the newest version 1.5.0 you can now select the default tool.

Please test and give feedback!

jaredarm, 2020-06-22:
Awesome! Thanks for getting back to me. I've changed the setting to tool1. I'll test it out today and let you know.

jaredarm, 2020-06-22:
The temperature now shows as expected. Nice work!
The filament details are still all zeroes though.

OllisGit, 2020-06-30:
Hi @jaredarm ,

please provide a gcode with multi-tool (or tool1) support, so that I can reproduce the behaviour, maybe the FilamentManager Plugin doesn't support the right values or read the wrong ones.

jaredarm, 2020-07-08:
Hey @OllisGit, apologies for the late response.

Attached is a copy of the gcode of a Marvin test print as well as an image of the post print page on the print history tab. All that I've changed is the image.

[MarvinWithImprovedCrotch-ABS-Medium.gcode.txt](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/4888429/MarvinWithImprovedCrotch-ABS-Medium.gcode.txt)

[image]

OllisGit, 2020-08-05:
Hi @jaredarm ,

the fix is implemented in the latest version 1.6.0.

Please test and give me a feedback.

Thx, in advance
Olli

jaredarm, 2020-09-14:
Apologies for leaving this for this long! I will test early this week and post the results.

jaredarm, 2020-09-15:
Hi @OllisGit,
I've just done another test print.
Here is the gcode: 
[MarvinWithImprovedCrotch-PLA-Fine.gcode.txt](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/5222677/MarvinWithImprovedCrotch-PLA-Fine.gcode.txt)

[image]
The calculated length is now showing but not the used quantities.

jaredarm, 2020-09-15:
I also just noticed that something strange is showing on the height "150.00 / 25.12mm"
It might be worth mentioning that I did update my installation to Python 3 recently.

OllisGit, 2020-09-19:
Hi @jaredarm,

you updated to Python 3, did you also update the FilamentManager-Plugin (V 1.6.0 works with Py3)? 
If you turn on the "dependency check" and reload the browser, then you can see if the filamentmanager is working (or look into the Plugin Manager Settings.

[image]

jaredarm, 2020-09-30:
Hi @OllisGit,

Yes, the Filament Manager is on V1.6 and I have the plugin dependency check option activated.

OllisGit, 2020-10-02:
Hi @jaredarm ,

which version of DisplayLayerProgress do you use, because in Version 1.23.0 (https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/tag/1.23.0) the evaluation of the height was changed.
In the newer versions the height is read from some octoprint-metafile, maybe there is the issue located

jaredarm, 2020-10-07:
Hi @OllisGit, 

I just checked, I'm on 1.23.3 of DisplayLayerProgress. I think the reason for the weird layer height is that I have some g-code at the end of a print that moves the print bed down (z goes up). I think it moves it 150mm so that would explain the 150.

The main issue is still that I'm not getting the correct values in the fields relating to the filament usage.

[image]

Let me know if there is anything I should try and re-installing or if there is anything I can send across to help with this issue. Thanks for your help so far.

closed 2020-11-16 by stale[bot]