OllisGit/OctoPrint-PrintJobHistory #63: [Request] Clone from previous

obraznori1722, 2020-07-01:
Awesome plugin, thanks for all the effort!

For those of us who do repetitive printing of exactly the same job (e.g., COVID shields/visors), it would be great to have a "Clone from previous job" and/or "Clone from another job" option, to automatically populate all the text fields from the previous/selected completed print job.

Either way, thanks again for the awesome plugin!

OllisGit, 2020-07-02:
Hi @obraznori1722 ,
nice idea! I will think about how a UI implementation can look like.

OllisGit, 2021-01-17:
Hi @obraznori1722,

I didn't found a good solution how I can include this feature into the UI, maybe you have an Iidea?

obraznori1722, 2021-03-02:
If I may suggest a simple solution: just a "Clone from last job" button, which would autopopulate (copy values from the last job's) text fields.

OllisGit, 2022-01-24:
Note to myself: create a simple dialog which shows all fields that are not "measured" with a checkbox in front of each. Each checked values is then copied to the new job.