OllisGit/OctoPrint-PrintJobHistory #64: "Something went wrong" attempting to upload image

M-Pixel, 2020-07-06:
> Something went wrong
> Maybe the filesize was to big (limit 5MB). Please consult octoprint.log for details

This appears whenever I try to upload an image.  It happens when the image is less than a single megabyte.  It happens when no image is actually selected at all.

`octoprint.log` receives only the following message when this occurs:

```
2020-07-06 15:51:47,953 - tornado.access - ERROR - 500 POST /plugin/PrintJobHistory/upload/snapshot/20200620-202152.jpg (10.147.19.162) 13506.08ms
```

`20200620-202152.jpg` is not the name of the file that I attempted to upload.  Perhaps `20200620-202152.jpg` is the very first image that I ever tried to upload, which probably was over 5MB.

How can I further diagnose?  Where is the 5MB limit defined?  Where might it be remembering `20200620-202152.jpg`?

OllisGit, 2020-07-22:
Hi @M-Pixel ,

I found the issue, there is a 5MB limit setting. Now I changed it to 20MB. 
This change will be in the next release (next days).

In the mean time you can change the filesize of the image (or delete it) with a "paint-programm".
The location of the snapshot folder could be found in the plugin-settings (tab: storage)

OllisGit, 2020-08-05:
Hi @M-Pixel ,

the fix is now implemented in the lastest version 1.6.0.

Please test and give feedback.
Olli

closed 2020-09-14 by stale[bot]