OllisGit/OctoPrint-PrintJobHistory #68: Feature request: Maintenance

MakerBirb, 2020-08-01:
Hello there, 

I had this idea to have the option to set up maintenance routines after certain amount of hours printing or after a set period of time. 
I presented my idea in the OctoPrint Discord server and jneilliii nudged me in your direction saying that it might be easy to implement since this plugin pulls print times and stuff anyway. 
My general idea was that you'd have several option to set up a routine which can be named and a description being added. Next step would be setting the 'timer'. So you could say that every 100 hours of printing a reminder pops up to tension the belts/lube the rods/check for wiggle or basically anything... 
Just an idea though :)

guysoft, 2020-08-01:
Hey, developer of OctoPi here,
The plugin has been unmaintaned, I just posted now that I would love to try and take control to maintain it.

MakerBirb, 2020-08-02:
Hello there!
So let's hope then it won't be unmaintained in the future :) I also hope you are considering my suggestion or if you know someone that would like to make a dedicated plugin, that'd be cool too :)

guysoft, 2020-08-02:
Hey, I started maintaing this plugin, but before I am considering new features I am bringing it up to date.

OllisGit, 2020-08-02:
Hi @guysoft !!!
> The plugin has been unmaintaned, I just posted now that I would love to try and take control to maintain it.

WHAT???? Are you talking about my Plugin "Print**Job**History" or "PrintHistory" from imrahil??

My release interval is between 10-20 days, so it is definitely **not unmaintained**.

OllisGit, 2020-08-02:
Hi @MakerBirb,

your mentioned request could look like this:

- Plugin-Settings: 
--  Maintenance-Intervall Timer: 100h (0h == deactivated)
-- Last Maintenance: 20.07.2020

- Each time a PrintJob is finished the plugin checks the maintenance Intervall against the last maintenance date. 
  If reached, a PopUp appears. The user must acknowledge this popup, if not the popup also appears after server startup 
  (I think this is needed if someone has the feature "shutdown after print" enabled) 

- After the user do the "real" maintenance he needs to update the date in the plugin - settings

Bonus-Level
- History of all maintenance with a note-field to document what was done/replaced.

Did I missed something?

guysoft, 2020-08-02:
PrintHistory, by bad, got my tabs mixed 😅

MakerBirb, 2020-08-03:
Hello @OllisGit,

My pleasure to meet you :) 
No I don't think you forgot anything important. Only thing might be a like say 3 month timer to check stuff like belts which are also affected by long exposure to being stretched. But I don't think it is most important. 
Thanks for the fast reply. And I hope we'll be seeing a cool update soon! Have a nice day

OllisGit, 2020-08-29:
Hi,

I am start playing around how I could implement this feature (with less effort). The current approach looks like this:

[image]

- User can activate the feature by entering a Begin-Date and a number of days. On that combination the next date for maintenance (MT) is calculated
- Everytime the user opens the OP-Page the plugin check if the next MT is reached.
- If reached, a popup appears with the hint the user must go to the settings to confirm the last MT
- The checkbox (maybe a button) could only be activated if a "Last MT" date was entered
- After activating the checkbox, the "Last MT" date is copied to the "Begin" field and "Last MT" is empty. next MT is calculated

Tell what do you think? Other ideas?

MakerBirb, 2020-08-31:
Hello!

Yes, indeed, this looks very promising. 
The only thing I would add - if possible - is a maintenance routine after a certain amount of hours printed. 
So let's say I want to check my belt tension every 100 hours of printing time. So after I printed stuff that in total adds up to 100h it reminds me to check belt tension the same way it does when I set it up with a date. 
Anyway so far it looks very good and to what you have so far I wouldn't have anything to add!

issackelly, 2021-02-22:
First, Thanks for all your work on this plugin! I truly appreciate the open source community, and your work specifically.

Second, I found this issue after digging around and some pointers here:
https://community.octoprint.org/t/a-wish-for-a-maintenance-log-for-my-printer/30982

I wanted to share that I was able to use the print history as a notebook with a little bit of lying. I think this will work plenty well for historical maintenance.

OllisGit, 2021-04-11:
fyi: It is planed: https://github.com/OllisGit/OctoPrint-PrintJobHistory/projects/1#card-44150656
It is in column long-term. Closed issue to get a better overview of "real" errors. It is still planed (not in column Done)

closed 2021-04-11 by OllisGit