OllisGit/OctoPrint-PrintJobHistory #72: The plugin is not working for me on the new OctoPrint 1.4.2

Murklock, 2020-08-11:
I recently update my OctoPrint to 1.4.2 and Print Job History stopped working completely for me. Do I need to completely uninstall and re install the plugin ? I look forward to hearing from you

OllisGit, 2020-08-15:
Thx, for testing with the newest OP release....I did not do it ;-)

OllisGit, 2020-08-15:
Hi @Murklock,

I updated to 1.4.2, but could not find any issue.

Please provide the octoprint.log file, maybe there is an error-entry.

Murklock, 2020-08-15:
Thanks so much for the reply ! Not a problem at all. I have added the log file below
[octoprint.log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/5079345/octoprint.log)

OllisGit, 2020-08-16:
Thx, for the logfile. 
There was an logging-error during collecting the filamentlength. This is not related to OP 1.4.2.
Fixed in newest version 1.7.0.

However...one thing is strange,  because the calculated filamentlength is provided by OP itself. It is stored in the ```.metadata.json``` file after upload.

Please take a look into that file for the failed print: ```VinylWallDisplayHolderV1OD.gcode```
The content should looks like this:
```
    "VinylWallDisplayHolderV1OD.gcode": {
        "hash": "750537bde6fd923612daec552e73e4813abcb1df",
        "analysis": {
            "printingArea": {
...
            },
            "dimensions": {
...
            },
            "estimatedPrintTime": 6676.9785850632,
            "filament": {
                "tool0": {
                    "length": 4229.171390000995,
                    "volume": 0.0
                }
            }
```
Thx, in advance
Olli

Murklock, 2020-08-18:
Totally works ! 

Thanks for looking at my log and guiding me in the right direction. I did look into the .metadata.json file but I couldnt find out what was going on so. Since you said it worked with the most recent version of Octoprint I went ahead and made a fresh image of Octoprint and installed all the plugins I needed and the plugin started working again without a problem. Thanks again so much for your help !

closed 2020-08-19 by OllisGit