OllisGit/OctoPrint-PrintJobHistory #74: [Feature request]: Timeframe statistics

OllisGit, 2020-08-19:
Most requested feature is "Statistic Report Diagram" (see https://github.com/OllisGit/OctoPrint-PrintJobHistory/issues/6), but implementing Diagrams takes a lot of effort (for me, I am not a JS/UI-Expert).

I decided to implement (in the future) the approach to send every event (that OP + Plugins create) is send to an "Influx-Database-Sink). See https://github.com/OllisGit/OctoPrint-PrintJobHistory/projects/1#card-27067834 

BUT: It should be possible to implement just a simple "timeslot picker" (start/end date and give  an result tableview of several aggregated values, like:

- Used Length/Weight
- Duration
- Filesize
- Used Material (Array)
- Used Spools (Array)
- Print result (count of each status)

osm-3d, 2020-08-19:
As I mentioned in a email to Olli the idea of having the ability to track total print time and filament length seems very appealing. Mock-up screenshots look much better than I imagined

OllisGit, 2020-09-19:
Implemented in Version 1.8+

closed 2020-09-19 by OllisGit