OllisGit/OctoPrint-PrintJobHistory #75: [Feature request]: Integrate PrintJobHistory-Plugin

OllisGit, 2020-08-19:
As the title says ;-)

Lumute, 2020-08-29:
Yes, +1 for this one, FilamentManager seems to be abandoned and no one has picked it up. SpoolManager 1.0 was just released and its looking great...

TDI4, 2020-08-30:
+1 on this, would be really nice. SpoolManager looks more detailed than Filament Manager. I would appreciate.

Thank you in advance!

cwbullet, 2020-09-08:
I would agree with this once Spoolmanager supports an external database.

ofc587a87, 2020-09-12:
+1 for me too. After migrating octoprint to Python3 (recomeded by octroprint blog), FilamentManager gets disabled, as it doesn't support python3.

cwbullet, 2020-09-12:
This has happened to me with Printer History, statistics 2.0, and now filament manager.

SM-26, 2020-09-19:
so far I prefer your sool manager over filament manager.
and print history integration seems like the next logical step.

If there is anything I could do to help, just say the word.

cwbullet, 2020-09-19:
I think if spool manager were able to have a central database for multiple printers/octoprints, I would switch tomorrow.

6ffm70, 2020-09-20:
+1 I thought you were to do that anyway, Olli! 

For this one, I'd love to see some filament recognition from the gcode, whereever possible - like discussed in [https://github.com/OllisGit/OctoPrint-PrintJobHistory/issues/58](url)

condraka, 2020-10-05:
++1 also.  Like SpoolManager better than Filament Manager.   
+2 for external database to use with multiple machines on different Octopi's running on those machines....

fabiradi, 2021-01-17:
There is a PR, which might be a good start (or even solve this issue completely?), see https://github.com/OllisGit/OctoPrint-PrintJobHistory/pull/102

OllisGit, 2021-04-11:
fyi: Currently I have some "private challenges to solve", thats the reason why It takes so long to release a new version, sorry!

Today I released a new development version (1.11.0). This version include the option for selecting the SpoolManager-Plugin for filament tracking.
**BUT you need the newest Version of SpoolManager 1.4.0+ and an other BUT: this version is not released yet!**

So, please be patient it will take some time.

Thx, in advance
Olli

OllisGit, 2021-05-24:
Try to use the lastest PJH-Plugin V1.14.0 and SpoolManger V1.4.0, that should work!

Please test and give feedback!
Olli

condraka, 2021-05-24:
Yes 

Carl E. Ondraka

closed 2021-10-30 by OllisGit