OllisGit/OctoPrint-PrintJobHistory #78: Pb with last version 1.8.0

jmarc67, 2020-08-30:
Hello,

1.7.0 was OK. With the 1.8.0 update, Intallation is OK but i have no more tab "print job history" and nothing about it in the octoprint plugin manager.

Reinstalled 1.7.0 and all is fine.

soerup, 2020-08-30:
Same here: 

2020-08-30 14:15:04,858 - octoprint.startup - INFO - Blacklist processing done, adding 10 blacklisted plugin versions: roomtemp (any), GcodeEditor (0.1.1), GcodeEditor (0.2.0), GcodeEditor (0.2.1), GcodeEditor (0.2.2), GcodeEditor (0.2.3), GcodeEditor (0.2.4), GcodeEditor (0.2.5), GcodeEditor (0.2.6), GcodeEditor (0.2.8)
2020-08-30 14:15:04,879 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2020-08-30 14:15:05,623 - octoprint.plugin.core - ERROR - Error loading plugin PrintJobHistory
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 972, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 71, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 27, in <module>
    from .DatabaseManager import DatabaseManager
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/DatabaseManager.py", line 16, in <module>
    from octoprint_PrintJobHistory.models.PrintJobSpoolMapModel import PrintJobSpoolMapModel
ImportError: No module named PrintJobSpoolMapModel

henrikba, 2020-08-30:
I also see the same -- tab gone after upgrade. I also don't see the plugin listed in installed (3rd party) plugins. I thus tried to re-install the plugin (after I took a backup of the data etc). Seems well-reproducible at this point, but let me know if you need log data or anything from me. Have captured and stored logs and screenshots.

tideline3d, 2020-08-30:
Same here, when I installed 1.8.0 the announcement popup that asks me to restart showed 'Unknown' as the plugin name.   Back to 1.7 working just fine (and all my settings/history remain in tact).

tideline3d, 2020-08-30:
@OllisGit You may want to temporarily pull this release out of the auto-update path as its not easy to roll back for an average user.

OllisGit, 2020-08-30:
SORRY, not my day!

The issue is fixed in latest version 1.8.1.

tideline3d, 2020-08-30:
Looks good to me.  I just cleared the updates cache on one of my instances, pulled the update, and everything appears to install fine.  I see the tab, my history and see it in the settings page.

I haven't tested any of the other stuff in 1.8 but the install issue looks fixed.

jmarc67, 2020-08-30:
Same for me. All is OK with 1.8.1

closed 2020-08-30 by jmarc67