OllisGit/OctoPrint-PrintJobHistory #80: Hit or Miss when capturing Used Length

crafttherapy, 2020-09-04:
Hi There... Love your plugin and am now working on a project that I want to track Used Filament. Sometimes its captures it, and sometimes it doesn't... 
[image]

I'm using the FilamentManager plugin and have a Prusa i3 MK2.5s MMU2s. Can you help me identify if its a hardware issue or a settings issue? I printed several jobs with the same spool and it was random as to when the PrintJobHistory plugin would capture Used Filament.

Thank you in advance.

OllisGit, 2020-09-19:
Hi @crafttherapy ,
the current implementation only captures the extrusion from one tool and the toolID must be selected in the Plugin-Settings.
Maybe in your case you choose a different tools and forgot the change the toolID in the settings.

BR
Olli

ArmedBastards, 2020-11-01:
Hi there!
Firstly, thanks for this absolute great plugin!

To me this also happens occasionaly - it looks like this always happens when a print has a pause in it.
The length then only seem to be captured from the last pause on.

Today I made a print with a pause on layer 17/20 to insert a magnet, and the print history plugin captured only 0,39m from  1,92m calculated (only 3 layers after the pause).
This was odd, because normally the captured value is near identical to the calculated one.

So I looked up my history of prints, and the captured length's were all near exact to the calculated length, apart from one print 2 weeks ago where I had a pause (filament change) on layer 8 from a total of 61 layers. Print History said the captured length is 3,02m from 3,90m calculated. (53 layers remaining after pause, so more length captured by the plugin).

All other prints show near identical values, only differing in 1 cm or so between calculated and captured length.

I add the pause in the print within cura with the Modify G-Code => "Pause at Height" plugin.
Maybe this helps to find the issue.

Greets
Lars

closed 2020-12-11 by stale[bot]

OllisGit, 2021-01-13:
Hi @ArmedBastards,
I created a new ticket: #111