OllisGit/OctoPrint-PrintJobHistory #81: Doesn't work anymore

masterdot, 2020-09-04:
Hey there,

My plugin doesn't track the prints anymore. Is there a problem or is my installation in a bad state?

OllisGit, 2020-09-05:
Hi @masterdot,
please provide the octoprint logs, maybe there is some error tracked.

masterdot, 2020-09-05:
It seems it does work again... Maybe a one time error?!

Enginecrafter77, 2020-09-06:
I experience the same issue. After the last update, PrintJobHistory doesn't log prints anymore. [Here](https://gist.github.com/Enginecrafter77/e3b6320505f79d67711203c9cbc3b90b) is the relevant octoprint.log data (filtered using `grep PrintJobHistory octoprint.log`).

The log says something about metafiles for the gcode files. I didn't study the codebase, but I blindly guess that the error might be that PrintJobHistory relies on uploading the gcode files through the website, but doesn't work for files uploaded through other means (for example FTP or SMB).

masterdot, 2020-09-06:
OK, now it doesn't work again... Added the log with your plug-ins on debug.
P. S. Spool Manager doesn't fill the filament Infos...

masterdot, 2020-09-10:
Here another log: https://www.dropbox.com/s/3dii3bbzvdzieqy/octoprint%20%282%29.log?dl=0

Spools are still not filled, but the last three prints did popup the history. Hope it does help a bit.

OllisGit, 2020-09-19:
Hi @masterdot ,

the used length is captured from "FilamentManager-Plugin" and this is disabled in your case.
[image]

OllisGit, 2020-09-19:
Hi @Enginecrafter77 ,

your case looks a little bit different, but I missed some informations in your look file:
[image]
The error is mentioned in Line 259, but there is no addition error-message, like "Object is None".

Please provide a full octoprint log.

BR
Olli

masterdot, 2020-09-19:
I have installed your spool manager and did disable the filament manager before I do delete it. So it seems to ignore your manager?

OllisGit, 2020-09-19:
Currently there is no integration between PJH and SpoolManager, its on my List.
The filament-values were only picked up from FilamentManager

masterdot, 2020-09-19:
Ahhhh... I thought it islready integrated because it's your plugin... That's the reason I did switch from filament manager to spool manager. To have the best integration bacause it's the same developer...

Enginecrafter77, 2020-09-19:
Hi @OllisGit, sorry for the late reply.

As I said, the error was caused because the gcode file was uploaded using smb directly to the uploads directory, and hence not analyzed properly (i.e. no metadata file was created). This was also reported by FilamentManager plugin. As such, uploading through the web interface solved this problem.

If I were to encounter this problem again, I will open up a new issue dedicated to this. But at the time of writing, the problem I had is resolved and no other problems persist. PrintJobHistory works exactly as expected, though I no longer use in combination with DisplayLayerProgress (because of unrelated reasons).

And sorry for incorrectly identifying the problem on my side, and thus posting in an unrelated thread.

Have a nice day.

closed 2020-11-03 by stale[bot]