OllisGit/OctoPrint-PrintJobHistory #83: Unable to install after migrating to Python 3

Lumute, 2020-09-11:
Hi,

I just tried migrating to Python 3 (3.7.3 specifically) but I'm unable to install this this plugin again. The installation does not give any error and then I get the message to restart but it says "Install Unknown" and then after the restart is like the plugin was never installed.

Here is the installation log:

2020-09-11 02:14:15,594 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2020-09-11 02:14:15,595 > Processing /tmp/tmpniam901o/master.zip
2020-09-11 02:14:17,559 > Requirement already satisfied: OctoPrint in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from Print-Job-History==1.8.1) (1.4.2)
2020-09-11 02:14:17,560 > Requirement already satisfied: pillow in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from Print-Job-History==1.8.1) (7.2.0)
2020-09-11 02:14:17,560 > Requirement already satisfied: peewee in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from Print-Job-History==1.8.1) (3.13.3)
2020-09-11 02:14:17,560 > Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (1.0.7)
2020-09-11 02:14:17,560 > Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (2.11.2)
2020-09-11 02:14:17,560 > Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (0.1.1)
2020-09-11 02:14:17,560 > Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (0.6.0)
2020-09-11 02:14:17,560 > Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (2.0)
2020-09-11 02:14:17,560 > Requirement already satisfied: feedparser<6,>=5.2.1 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (5.2.1)
2020-09-11 02:14:17,561 > Requirement already satisfied: tornado==5.1.1 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (5.1.1)
2020-09-11 02:14:17,561 > Requirement already satisfied: watchdog<1,>=0.10.2 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (0.10.3)
2020-09-11 02:14:17,561 > Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (0.57.0)
2020-09-11 02:14:17,561 > Requirement already satisfied: future<1,>=0.18.2 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (0.18.2)
2020-09-11 02:14:17,561 > Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.06.22 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (2020.6.22)
2020-09-11 02:14:17,561 > Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (1.1.0)
2020-09-11 02:14:17,561 > Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (5.3.1)
2020-09-11 02:14:17,561 > Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (2.8.5)
2020-09-11 02:14:17,561 > Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (1.2)
2020-09-11 02:14:17,561 > Requirement already satisfied: regex!=2018.11.6 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (2020.7.14)
2020-09-11 02:14:17,562 > Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (1.5.0.1)
2020-09-11 02:14:17,562 > Requirement already satisfied: blinker<2,>=1.4 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (1.4)
2020-09-11 02:14:17,562 > Requirement already satisfied: rsa==4.0 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (4.0)
2020-09-11 02:14:17,562 > Requirement already satisfied: OctoPrint-FileCheck>=2020.07.06 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (2020.8.7)
2020-09-11 02:14:17,562 > Requirement already satisfied: pylru<2,>=1.2 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (1.2.0)
2020-09-11 02:14:17,562 > Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (2.24.0)
2020-09-11 02:14:17,562 > Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (1.12.1)
2020-09-11 02:14:17,562 > Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (3.1.1)
2020-09-11 02:14:17,563 > Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (1.1.1)
2020-09-11 02:14:17,563 > Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (0.10.9)
2020-09-11 02:14:17,563 > Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (0.17.4)
2020-09-11 02:14:17,563 > Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (0.4.21)
2020-09-11 02:14:17,563 > Requirement already satisfied: psutil<6,>=5.7 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (5.7.2)
2020-09-11 02:14:17,563 > Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (3.4)
2020-09-11 02:14:17,563 > Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (7.1.2)
2020-09-11 02:14:17,563 > Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (1.1.2)
2020-09-11 02:14:17,564 > Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (0.5.0)
2020-09-11 02:14:17,564 > Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (0.8.0)
2020-09-11 02:14:17,564 > Requirement already satisfied: sarge==0.1.5post0 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (0.1.5.post0)
2020-09-11 02:14:17,564 > Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (1.0.0)
2020-09-11 02:14:17,564 > Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.8.1) (1.0.1)
2020-09-11 02:14:17,564 > Requirement already satisfied: webassets>=2.0 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from Flask-Assets<3,>=2.0->OctoPrint->Print-Job-History==1.8.1) (2.0)
2020-09-11 02:14:17,564 > Requirement already satisfied: pathtools>=0.1.1 in /home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages (from watchdog<1,>=0.10.2->OctoPrint->Print-Job-History==1.8.1) (0.1.2)
[... cut ...]
2020-09-11 02:14:18,906 > Stored in directory: /tmp/pip-ephem-wheel-cache-o8iley2l/wheels/4e/c4/a7/07ef33c91773e55904289ffcda94ec7523c3ef6fd493a7f2d0
2020-09-11 02:14:18,906 > Successfully built Print-Job-History
2020-09-11 02:14:18,906 > Installing collected packages: Print-Job-History
2020-09-11 02:14:18,906 > Attempting uninstall: Print-Job-History
2020-09-11 02:14:18,906 > Found existing installation: Print-Job-History 1.8.1
2020-09-11 02:14:18,906 > Uninstalling Print-Job-History-1.8.1:
2020-09-11 02:14:18,907 > Successfully uninstalled Print-Job-History-1.8.1
2020-09-11 02:14:18,907 > Successfully installed Print-Job-History-1.8.1

OllisGit, 2020-09-13:
Please look into the log after the server is restarted.
You should find something like this: 
```
...
|  PrintJobHistory (1.8.1) = /Users/o0632/0_Projekte/3DDruck/OctoPrint/OctoPrint-PrintJobHistory/octoprint_PrintJobHistory
...
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
2020-09-13 11:28:46,216 - octoprint.environment - INFO - Detected environment is Python 3.6.8 under Macos (darwin). Details:
|  hardware:
|      cores: 8
|      freq: 2300
|      ram: 17179869184
|  os:
|      id: macos
|      platform: darwin
|  python:
|      pip: 20.2.2
|      version: 3.6.8
|      virtualenv: /Users/o0632/0_Projekte/3DDruck/OctoPrint/octoprint_latest/venv-3-op142
```
Maybe there is some prefix in front of the PrintJobHistory-Entry (```!, #, *```).

As you can see, I am using Python 3.6.8., so will I will try to update to 3.7.X

Lumute, 2020-09-14:
This is what I get in the log during startup:

2020-09-13 23:32:09,466 - octoprint.plugin.core - ERROR - Error loading plugin PrintJobHistory
Traceback (most recent call last):
  File "/home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages/octoprint/plugin/core.py", line 972, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages/octoprint/plugin/core.py", line 71, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 241, in load_module
    return load_package(name, filename)
  File "/home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 215, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 25, in <module>
    from .api.PrintJobHistoryAPI import PrintJobHistoryAPI
  File "/home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 23, in <module>
    from octoprint_PrintJobHistory.api import TransformPrintJob2JSON
  File "/home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/TransformPrintJob2JSON.py", line 4, in <module>
    from octoprint_PrintJobHistory.CameraManager import CameraManager
  File "/home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages/octoprint_PrintJobHistory/CameraManager.py", line 9, in <module>
    from PIL import Image
  File "/home/pi/OctoPrint/oprint-env/lib/python3.7/site-packages/PIL/Image.py", line 94, in <module>
    from . import _imaging as core
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory

Lumute, 2020-09-14:
Ok, I didn't have libopenjp2-7 installed so after installing it the plugin loaded just fine:

sudo apt-get install libopenjp2-7

I guess this not a requirement on python 2? weird...

OllisGit, 2020-09-14:
Hi @Lumute,
thanks for investigation!

It looks like that the lib I need for the image processing (PILLOW) is available in two versions:
```python-pil``` and ```python3-pil```.
But I need to setup a clean environment to find out what is the minimum required-libs I need to configured in OP-Settings.

Maybe there is also a gap between different Python3-Versions.

OllisGit, 2020-09-19:
fyi: In latest version 1.9.0 I changed the lib version of pillow (same as Octolapse). This should fix the issue with Python 3 installation.
```
>=6.2.0<7.0.0", # since 7.0.0 no Python 2.7 Support, see https://github.com/python-pillow/Pillow/blob/master/CHANGES.rst
```

closed 2020-09-19 by OllisGit